arendst/Tasmota #1001: Feature request: RF Bridge - unlimited (?) codes and possibility for full code manual input

sgofferj, 2017-10-10:
Similar to ESPurna, it would be nice if Tasmota would support an unlimited - or limited by available memory - amount of codes for sending and transmitting.

It would also be nice if it was possible to not only learn a code but also enter one manually. The latter would be useful for easy addition of code-learning devices which don't come with its own transmitter, such as RF alarm strobes, sirens, etc. One could simply make up a code oneself and let the device learn that. The input field should support the input of the full code - including the "house code" prefix where applicable.

A cmnd/sendKey MQTT option would be the caking on the ice :)

MatzeJoerling, 2017-10-16:
+1 
i would appreciate such a feature!

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst

Tresett, 2017-11-28:
Sorry for dummy question
Can I send this command from Domoticz via MQQT?