arendst/Tasmota #1002: Blocked Loop on "reset 2"

stefanbode, 2017-10-10:
On the nodemcu and wemos the watchdog bites on erasing the flash (3MB). There is an additional line required in settings.ino line  312 to update the osw_last_loop .

```
  for (uint32_t _sector = _sectorStart; _sector < _sectorEnd; _sector++) {
    noInterrupts();
    result = spi_flash_erase_sector(_sector);
    interrupts();
    if (_serialoutput) {
      Serial.print(F(D_LOG_APPLICATION D_ERASED_SECTOR " "));
      Serial.print(_sector);
      if (SPI_FLASH_RESULT_OK == result) {
        Serial.println(F(" " D_OK));
      } else {
        Serial.println(F(" " D_ERROR));
      }
      delay(10);
    }
    osw_last_loop = millis();
  }
```

arendst, 2017-10-10:
Thnx. In next (pre)release.

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst

ITully, 2018-04-08:
turn off all logging