arendst/Tasmota #1003: Switch Causes Son off Basic to "Flicker" on GPIO - High

Paulf007, 2017-10-10:
Not sure if it is a HW or SW issue. 
I have a Son off Basic with a additional relay connected to GPIO14 - Relay2
Then I have two household toggle light switches connected to GPIO3 - Switch1 and GPIO1 - Switch2
When either of the two Switch modules are in a HIGH state the their is a constant flicker of the light you can also see in the console that the switch goes on/off 
"
17:12:49 MQT: stat/lounge_switch/RESULT = {"POWER1":"ON"}
17:12:49 MQT: stat/lounge_switch/POWER1 = ON
17:12:49 MQT: stat/lounge_switch/RESULT = {"POWER1":"OFF"}
17:12:49 MQT: stat/lounge_switch/POWER1 = OFF
17:12:50 MQT: stat/lounge_switch/RESULT = {"POWER1":"ON"}
17:12:50 MQT: stat/lounge_switch/POWER1 = ON
17:12:50 MQT: stat/lounge_switch/RESULT = {"POWER1":"OFF"}
17:12:50 MQT: stat/lounge_switch/POWER1 = OFF
"
If the GPIO is pulled to low ( Switch is on so GPIO to Ground) you do not get the Flicker and the switch runs stable even if I switch the Relay of via MQTT and the GPIO state is still low. 
Am I missing something? As I understand it the switch module will toggle the relays on the change of the corresponding GPIO port from low to high or high to low
Regards 
Paul

davidelang, 2017-10-11:
hook up a 5-10kohm resistor between the GPIO and 3.3v, that will probably solve 
the problem. If not, add a small (pF range) cap between the GPIO and ground.

Paulf007, 2017-10-11:
Ah you see that is where the lack of electronic knowledge come to one's detriment. Have placed a 10k and it seems to be perfect now! Thank you.
@davidelang would you mind explaining why there needs to be a resistor or what I should google for ?

davidelang, 2017-10-11:
search for pull-up resistor.

there is supposed to be one in the chip, but it doesn't seem to work very well.

What's happening is that you put the pin in the input mode, and when you close 
the switch, it goes to ground. But when the switch is open, what is there to 
make it read 3.3v?

a pull-up resistor solves this problem.

David Lang

closed 2018-05-07 by stale[bot]

MightyMilster, 2018-09-10:
Hi, I also had ghost-switching. Tried adding a pull-up resistor. Replaced old sonoff with a new sonoff. Set Tasmota to "PowerRetain - On". None of this helped and the blinking continued.

I eventually solved my problem and thought I'd add it here in case it helps someone else going crazy like I was. Remember that we are dealing with controlling a high voltage device. When you add a manual switch to GPIO14, you are dealing with a low voltage. This low voltage is usually being carried on very thin wire. What happens when high voltage cables are placed near low voltage cables - Induction!!!! Mysterious unpredictable ghost switching. I suggest keeping the two as far apart as possible and definitely use shielded cable for your switch wires. Good luck!