arendst/Tasmota #1005: sensor modification/ configuration

chris-008, 2017-10-10:
Hi,
Since the Itead original version of the Si7021 (none bare one) is not compatible yet with the TH10/16 sonoff Tasmota modules (MQTT based),  I am looking for an alternative.
The BME280 is on the list of supported sensors (https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration). Being one of the better sensors, I was thinking of buying this.

A few questions come up: 
1.	 Is it possible to modify the Si7021 sensor with the new BME280 PCB? Therefore spossible of soldering the existing jack/ wires  on the new BME280?

[image]
[image]

2.	Can the TH10 module handle this extra pressure parameter (or should it be some kind of TH”P” module instead)  next to the temp and humidity?
3.	GPIO configuration as in the screenshot underneath?

[image]

4.   Looking at the I2C sensor overview, what is meant with "group 3a” or groups in general? 

thanks for your replies!

mkh595, 2017-10-11:
Add 2) Pressure will be displayed automatically

[image]

mkh595, 2017-10-11:
add 1) It is possible to change Si7021 with BME280 but not with this from your picture, because it is sensor with SPI protocol. You have to look for I2C version, e.g.
[image]

chris-008, 2017-10-16:
Thanks for your info mkh595 , will buy the i2c chip  as suggested!

chris-008, 2017-11-07:
Hi again, I received the sensor as suggested. 
Looking at the pinout now of my original Itead SI7021, I only have a VCC, ground and data wire. I assume that I can not reuse the SI7021 casing and jack connector for this new print.
Should I buy a jack with 4 wires or perhaps solder it directly to the TH10 module?

[image]