arendst/Tasmota #1006: Feature request:  Hostname on startup webpage

kfonda, 2017-10-10:
I would like to see the Hostname on the first webpage seen when you connect to the device.

Like This:

[image]

ionciubotaru, 2017-10-11:
You can see the hostname in the browser's address bar

kfonda, 2017-10-11:
@ionciubotaru That's a great point, but I was accessing them by ip address and it only shows that in address bar. It's not really important since I can get the info from the Information screen, bit it would just make it one less step.

ascillato, 2018-03-06:
Hi @jonatanolofsson, The PR #2091 does not work. Sorry.

By default, the Hostname value is `%s-%04d` so, using `Settings.hostname` is not the best option.

[image]

To be displayed as @kfonda wants, should be: `my_hostname` with ".local"

Also can be added the IP address like **SOnOff_123.local (192.168.1.101)** if you like.

All this info can be reached also on Information Menu.

ascillato, 2018-03-06:
Tested the @jonatanolofsson approach with the review changes and work.

[image]