arendst/Tasmota #999: Tasmota with Google Assistant

nunoronconcouto, 2017-10-10:
Hi!!

Is possible to connect sonoff with tasmota firmware to Google assisant???

nunoronconcouto, 2017-10-10:
Is possible say say to my phone: Ok google, turn of bedroom light ( sonoff with tasmota firmware), because i have one sonoff pow ( original) with ewelink and work nice.
i have 4 sonoff and 4 sonoff touch ( all with tasmota for work with Home Assistant)

jesusvallejo, 2017-10-10:
its pretty difficult to understand u , it is posible if you have the google home speaker , enabling the hue emulation, but with out it i couldnt get the google home app to find my sonoffs emulating hue or wemo.

nunoronconcouto, 2017-10-10:
Sorry...
I have GooglePI ( google assistant on rp) and google assistant on my phone. i have one sonoff pow with original firmware work well because use ewelink. My 4 sonoff and 4 sonoff touch  with tasmota firmware, work with Home Assistant, is possible to use tasmota firmware with Google Assistant? 
When on my phone open google assistant setting/home control and try add new device...i try to link service philips hue ..say cant find ...

sorry my english

jesusvallejo, 2017-10-10:
GooglePi is not prepared for discovering devices, as i told you in the previus post, the only way you can make it work with your actual config, is using openhab.

nunoronconcouto, 2017-10-10:
OK...GooglePi is not prepared for this..OK ...but my phone is prepared...i link ewelink account and xiaomi  Yeelight lights

jesusvallejo, 2017-10-14:
yeah your phone is prepared , but only with oem firmware, hue emulation only works if u have a phisical devide such as alexa or google home mini, max or original

leranp, 2017-10-15:
I have tried to connect with google home and it doesn't work at all, it isn't doesn't find the device.

I have searched in the internet for solution and i so that peoples say that google ot Philip block this option and now it doesn't work for most of the peoples

jesusvallejo, 2017-10-15:
yeah it looks like is no longer suported,https://github.com/bwssytems/ha-bridge

> ISSUE: Google Home now seems to not support local connection to Philips Hue Hubs and requires that it connect to meethue.com. Since the ha-bridge only emulates the local API, and is not associated with Philips, this method will not work. If you have an older Google Home application, this may still work. YMMV.``

timmo001, 2017-10-16:
Not sure how it is still working with what I use then. I have all my Sonoff devices set up with Home Assistant and have no issues using the emulated_hue component

brookb52, 2017-10-18:
Hi, I am having the same problem with Hue Emulation on Sonoff with Tasmota. 
Here is a list of my components/setup.
Google home speaker (physical).
Openhab 2.1, MQTT running on raspberry pi 3B.
Multiple Sonoff Pow with tasmota firmware 5.8.0j hue emulation enabled
1 Sonoff Switch with tasmota firmware 5.8.0j hue emulation enabled
A Philips Hue Bridge with 6 hue white bulbs connected and assigned to rooms.
Openhab works flawlessly controlling the hue bulbs (via the openhab hue binding) and also controlling all of the sonoff devices (via MQTT). Google Home will find/control hue devices if they are available in the Hue app shown on my Android Device (which is linked to myhue online service). ie google home tells myhue to control my hue bulbs. All good and working well.

What I want to do is to also control my sonoff devices using google home via the sonoff tasmota hue emulation feature. Therefore I need to setup my sonoff hue emulated devices on the Hue android app.
I can add the emulated devices to the app as bridges but when I try to allocate the switch to a room (required by the hue app) the hue app responds with "Can't create default scenes because your Hue bridge is full. Try deleting some routines or scenes first." the app will not go any further and I am unable to control or find and therefore contorl the sonoff hue emulated devices with my google home app.
I am asuming that the hue app is trying to write to the sonoff device which of course is not possible. 
does anyone know how to get this to work?

altelch, 2017-10-18:
Hi,
> What I want to do is to also control my sonoff devices using google
> home via the sonoff tasmota hue emulation feature. Therefore I need to
> setup my sonoff hue emulated devices on the Hue android [...]

that's the point. If it's required to write scenes or room it can't 
work. The HUE emulation is very minimal in implementation and there is 
no support for groups or rooms or scenes. The API accepts the request 
but just emulates minimum requirements. See 
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_wemohue.ino 
about line 712. Groups have a handler, scenes just have the 
todo-handler. Groups handler ist about line 666, todo at 433 with not 
implemented answer. That's why your App can't create scenes or groups. 
If this is required for Google Assistant to work you can't get this 
running. If Google Assistant requires Hue Cloud things get worse as this 
will never work. The groups/scenes could be written but I'd guess it 
would take to much space on the limited sonoff ressources.

Greetings,

          Heiko.

brookb52, 2017-10-18:
Thanks Heiko. Understood.  Amazon echo is looking really good right now, from what I gather the echo supports local hue emulation.

davidelang, 2017-10-18:
instead of trying to get google assistant to talk directly to the devices, how 
about having it talk to openhab? that should give you far more power and 
flexibility than talking to the devices directly.

Mar-85, 2017-11-23:
Im wondering about this too, the sonoff OEM firmware now supports google home right out of the box! Would be great if this did also.

MikePRoberts, 2017-11-26:
There are solutions here.  However don't expect any Sonoff-Tasmota device to communicate direct to Google Home.  The firmware here communicates only with an MQTT broker using topics and payloads as shown in the Wiki.  So the question needs to be how to connect Google Home to your MQTT broker - which arguably is not a topic for for this site.  The solutions are not dependent on Tasmota but on the broker and/or another device/system that might be connected to the broker.  One option is to find a system to listen to Google Home and send commands either direct to your broker or possibly to a system like Node-Red (on a Raspberry Pi) and configure Node-Red to send the appropriate topics/payloads to your broker. Some searches on "Google Home MQTT" "Google Home 'your home automation system' " "Google Home Node-Red" might lead you so a solution that will work on your system.  Also note that Google Home relies on an internet connection to work.  Hence think about what happens when the internet is down and whether you are happy with the system security.
I hope this helps.  Mike

brookb52, 2017-11-26:
I believe the original question refers to the tasmota hue emulation which is certainly a topic for this thread and Mqtt is certainly not the only communication that is supported. Ie http, rest, mqtt, and both wemo and hue emulation. That said thanks for the info re node red I'll check it out when I get a chance.

MikePRoberts, 2017-11-26:
My apologies. I should have considered the other coms options.  In case it helps - I am using Node-Red and Mosquitto (broker) on a Raspberry Pi that is nice as it sits entirely on my local network.  Node-Red however has a load of other functionality in addition to MQTT nodes that make it a useful system for home automation.  However is is just another MQTT client and so could be used with any MQTT based automation system. See: 
https://www.instructables.com/id/Powerful-Standalone-Home-Automation-System-Pi-Sono/
for the Node-Red/Mosquitto setup and : 
https://www.instructables.com/id/Home-Automation-Sonoff-Tasmota-Sensors-LEDs-Develo/
for driving sensors and LEDs.
Mike

joseant, 2017-11-30:
I have managed to work with this combination:
1º With IFTT you configure the google asisstant as  trigger 
2º With IFTT you configure the action as webhooks that go to http: // yourddns: port / cm? Cmnd = Power% 20On
3º in the router do NAT from the port to the ip of the device.

  you should add the user and password in the device

Jswee1, 2017-12-03:
@joseant I've tried doing this but it doesn't work I've port forwarded the sonoff then put my ip in the web hook  example you gave the response comes up in assistant but nothing happens with the sonoff so i believe its configuration with the web hook. Is there any other options i need to change for the web hook?

marcinbauer85, 2018-01-01:
I would also want to know if it's possible to use Tasmota/(other alt fw), and Google Home speaker to control sonoffs. Right now Im testing my Sonoff with Tasmota + Hassio, but my speaker isn't connected to all of that :(

profesionalTI, 2018-01-01:
Hi, try with node-red is great only you need search Google Home module to install search in Google: node-red google home

Coordialmente,
Best Regards,

Marcelo A. Anuch Medina
Consultor de Empresas Público/Privadas
Ingeniero Comercial, MBA UDD
Business Management

🔒   La información contenida en esta transmisión es confidencial y está dirigida exclusivamente a el o los destinatarios indicados. Cualquier uso por otras personas que no sean sus destinatarios está sancionado por la ley chilena. Si ha recibido esta transmisión por error, por favor destrúyala y notifique al remitente si corresponde. Las opiniones expresadas en este correo son las de su autor y pueden no ser compartidas por la compañía. Marcelo A. Anuch Medina,  no asume obligaciones ni responsabilidades por el contenido de este correo. Tampoco garantiza que la transmisión de este correo sea segura o libre de errores, y en consecuencia, no responde por virus, pérdidas, destrucción, recepción tardía o incompleta.

🍀  Antes de imprimir este correo electrónico asegúrese que sea realmente necesario: Cuidemos el medio ambiente! Todos vivimos en un solo planeta.

marcinbauer85, 2018-01-01:
But I want to use the original Google home speaker that I bought from Google, not to make a port of it on a PI. And from what I understand, the google Home App doesn't have anything that I can add as a device when I flash a sonoff with Tasmota/Espeasy/etc, because Sonoffs are only supported by GH with the eWeLink app.

joealmonte, 2018-04-25:
It is now possible to integrate Google Assistant with Home Assistant and Tasmota Sonoff devices. Follow the link to BurnsHA instructional YouTube video.

https://www.youtube.com/watch?v=Xk0CafW4_Pg
https://www.youtube.com/watch?v=iS_88VzrFuw

The second link is the updated method, but please take a moment to watch the first link so you have a basic understanding on how it used to work before the updated method.

brookb52, 2018-06-09:
So in short the answer to the initial question is no not directly. I personally use openhab, mqtt and the openhab plugin for google to control tasmota with my voice.

ascillato, 2018-06-13:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2

mikizvi, 2018-12-08:
Hi,

Just pointing out that [Kappelt gBridge](https://about.gbridge.kappelt.net/) lets you bridge MQTTs.  More detailed comment [here](https://github.com/arendst/Sonoff-Tasmota/issues/2146#issuecomment-445438821).

Michael.

harkoy, 2019-02-16:
Hi, you can use Yonomi (Google store app) it recognizes sonoff tasmota (Hue mode) and then you can add yonomi (and their configured devices) to Google Home like any other device manager.

dony71, 2019-04-23:
@harkoy, yonomi can detect sonoff tasmota in hue mode emulator but when i add yonomi in google assistant, it cannot see the device. this means, i cannot control sonoff tasmota via voice on google home. how to do properly?

harkoy, 2019-04-23:
did you try setting up a scene in yonomi? it should appear in google's devices list then

dony71, 2019-04-24:
tried scene in yonomi and voice instruction on google home can trigger yonomi scene and pop up notification. however sonoff does not respond to switch on/off.