arendst/Tasmota #1007: Sonoff dual, does not upgrade fw via web

Tresett, 2017-10-11:
Can't upgrade firmware on Sonoff Dual via web.

1. Working on ESPEasy. Download from github (Tasmota) already compiled sonoff.bin and try to update via webform. Failed and not back to ESPEasy firmware.
2. Ok. Compile Tasmota project in Arduino IDE, load firmware to device via serial port. All working.
3. Add few changes in Tasmota project (little changes un user_config.h), compile in Arduino IDE. Try to update fw via web. Failed and not back to working mode. Button not working too.
4. Today i'm try to upload fw via serial.

Do I smth wrong?

Tresett, 2017-10-26:
TH16 has same problem
May be firmware is too big, for upload via web?
I'm purchase new chips for upgrare memory 1M->4M

Link to describe few hacks: https://www.jackenhack.com/sonoff-switch-i2c/

davidelang, 2017-10-26:
did you try uploading the minimal version? it has most features disabled, but 
exists so that you can load it, and then load a larger image.

Tresett, 2017-10-27:
Compile and upload minimum version. Suscesseful.
But in minimum version Web Management is off. 9-(
How to upload new firmware? OTA? I have not expireience to configure and start update OTA via MQQT =)

I'm disable all unneeded options, recompile it, new firmware is -20kb.
Upload successeful, I'm happy =)

P.S. But chips already purchased.

closed 2017-11-29 by Tresett