arendst/Tasmota #1008: Default button behaviour

mczac, 2017-10-11:
What is the best way to change default button behaviour on Sonoff DUAL?
I need it to implement simple behaviour when closed(button pressed) all relays (1,2) on, when opened relays (1,2) off.
Thanks!

davidelang, 2017-10-11:
the firmware does not support this directly, you would need to set the 
buttontopic so that pressing the button does nothing directly, but sends amqtt 
message to your home automation server, that server would then need to send 
instructions to turn on both relays.

mczac, 2017-10-11:
Thanks.   
  
  

 Does it send 2 mqtt messages? When I Press and when I release?
  
I tried setting up a button topic, but firmware behaviour kicks in when holding the button for reset.

davidelang, 2017-10-11:
I think you are going to have to define a new module type where the button's 
GPIO is defined as a switch, then you can use switchmode to set it to follow the 
switch and send a message on connect and disconnect.

mczac, 2017-10-11:
Thanks David, I don't seem to find built-in button defined in the profile
is it wired directly to the ESP or it works with serial controller?
Other option could be to use GPIO13 and remove the LED

{ "Sonoff Dual",     // Sonoff Dual (ESP8266)
     0,
     GPIO_TXD,         // GPIO01 Relay control
     0,
     GPIO_RXD,         // GPIO03 Relay control
     GPIO_USER,        // GPIO04 Optional sensor
     0,
     0, 0, 0, 0, 0, 0, // Flash connection
     0,
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Optional sensor
     0, 0, 0
  },

tom211, 2017-10-11:
I have looked into the [schematic](https://www.itead.cc/wiki/images/1/1d/Sonoff_Dual_View.pdf) The Button is connected to the additional controller. I think there is no possibility to change its behaviour...

closed 2017-10-11 by mczac

davidelang, 2017-10-12:
not without code changes :-(

That said, you do have the code...