arendst/Tasmota #1009: Multiple DS18B20 extract data Hassio

vicok, 2017-10-11:
Hi...need some help. Installed an mqtt sensor Tasmota on D1 mini. Seems to transmit the status into Hassio. State shows {"Time":"2017-09-28T18:15:50", "DS18x20":{"DS1":{"Type":"DS18B20", "Address":"28FFDAE770170468", "Temperature":26.7}, "DS2":{"Type":"DS18B20", "Address":"28FFCEBA7017044E", "Temperature":27.9}}, "TempUnit":"C"}. What should I put into config.yaml to extract the temp values?
Can't get it working

mkh595, 2017-10-11:
value_template: "{{ value_json.DS18x20.DS1.Temperature }}"

vicok, 2018-04-28:
> Just in case you're like me and want to extract value by sensor address,
> e.g.:
>
> value_template: "{% for key, value in value_json.DS18x20.items() if
> value.Address == "28FFDAE770170468" %} {{ value.Temperature }} {% endfor %}"

thx Piotr

closed 2018-06-27 by stale[bot]

ppnacho, 2019-08-04:
> Just in case you're like me and want to extract value by sensor address, e.g.:
> 
> `value_template: "{% for key, value in value_json.DS18x20.items() if value.Address == "28FFDAE770170468" %} {{ value.Temperature }} {% endfor %}"`

Hi, I have an error when I use your code, my config,

sensor hexid4:
  platform: mqtt
  name: "00000B10C80F Temperature"
  state_topic: "tele/pbaja1/SENSOR"
  unit_of_measurement: "°C"
  value_template: "{% for key, value in value_json.DS18B20.items() if value.Address == "00000B10C80F" %} {{ value.Temperature }} {% endfor %}"
  payload_available: "Online"
  payload_not_available: "Offline"

and the error,

can not read a block mapping entry; a multiline key may not be an implicit key 

Please, could you help me. Thanks