arendst/Tasmota #1012: How to config 2 pushbuttons for one relay

eimix, 2017-10-12:
My goal is to use sonoff switches combined with wall switches.

Tried using SwitchMode1/SwitchMode2 with simple wall switch (togle button), but it is not so "self explaining" to home users if button is left in ON position and then light is turned OFF remotely with MQTT (OpenHAB2).

Is it possible to config sonoff basic with connected 2 pushbuttons?
[image]
(this image shows two keys(left/right) each with 2 pushbuttons (top/bottom))

Wanted action:
Top pushbutton always switch relay ON
Bottom pushbutton always switch relay OFF.

ionciubotaru, 2017-10-12:
check the wiki for `switctopic` and `switcmode`

eimix, 2017-10-12:
I checked it multiple times, found undocumented SwitchMode5/6 - but no success.

SwitchMode0
SwitchMode3
SwitchMode4
works with pushbuttons - but it always toggles relay if pressed multiple times.
In my case - i do not need to toggle relay, if "top" pushbutton is pressed twice or more. Relay must stay ON unless bottom pushbutton is pressed.

SwitchMode1
SwitchMode2
does not work with pushbuttons - as relay folows the state of pushbutton.

davidelang, 2017-10-12:
On Thu, 12 Oct 2017, eimix wrote:

> SwitchMode1
> SwitchMode2
> does not work with pushbuttons - as relay folows the state of pushbutton.

This can work, you use your home automation software here instead of managing 
the relay directly.

you set switchtopic so that these send mqtt messages to your home automation 
server, and it ignores the contents of the message, you just have it watch for a 
message from button1, and when it gets one, it sends a message to turn relay1 
on. When it sees a message from button2, it sends a message to turn relay1 off

eimix, 2017-10-12:
It looks like solution,
but it would not work if wifi network is lost or mqtt or openhab2 is down. 
it would be nice to have direct relay control.

davidelang, 2017-10-12:
there's a limit to how many special cases can be put into the firmware, it's got 
a very limited amount of flash, and we are already at the point where we cannot 
update if all features are enabled due to this limit.

Is this worth loosing some other feature for?

Remember, you do have the source code, so you can add features that you want.

David Lang

eimix, 2017-10-12:
I checked source,
found easy way to modify it for my self in sonoff.ino 
"switch (sysCfg.switchmode[i]) {"
with new switchmode7/8, but did not knew about "size" issue.

before any changes wanted to check, if this could be solved with config.

thank you for your tips.

closed 2017-10-12 by eimix