arendst/Tasmota #1013: Support a command to configure whether Tasmota sets the MQTT retain flag on ENERGY messages

iangregory, 2017-10-12:
It would be useful if there was an `EnergyRetain` command that allowed the MQTT retain flag to be sent on messages such as this one:

```
fridge_freezer/tele/ENERGY {"Time":"2017-10-12T10:00:16", "Total":4.131, "Yesterday":0.637, "Today":0.244, "Period":1, "Power":95, "Factor":0.79, "Voltage":242, "Current":0.496}
```

This would allow dashboard apps to display the most recent values as soon as they were opened, rather than waiting for the next ENERGY message.

arendst, 2017-10-12:
Will be in next (pre)release as an inclusion of command ``SensorRetain``

mkh595, 2017-10-17:
Retain flag is not always good solution. Later you may recognized also disadvantages.
If Sonnof will be disconnected due to wifi off, power off, failure, stollen :-), ..., your dashboard will still show you last value and you will not recognized that something is wrong.

iangregory, 2017-10-17:
@mkh595 The message contains a timestamp so it's easy for the dashboard or monitoring tools to flag stale values, especially with short telemetry intervals

@arendst Thanks!

closed 2018-01-03 by iangregory