arendst/Tasmota #1014: Feature Request: update IRremoteESP8266 to v 2.x and add support for the panasonic protocole

stylab, 2017-10-12:
First thanks for the great work you are doing

I am hoping that you will update IRremoteESP8266 to v 2.x and add support for the panasonic protocole

arendst, 2017-10-13:
Could you please test Panasonic functionality with current library by making the following addition to file xdrv_ir_send.ino around line 324:
```
          else if (!strcasecmp_P(protocol, PSTR("SAMSUNG")))
            irsend->sendSAMSUNG(data, bits);
          else if (!strcasecmp_P(protocol, PSTR("PANASONIC")))
            irsend->sendPanasonic(data, bits);
          else {
            snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_IRSEND "\":\"" D_PROTOCOL_NOT_SUPPORTED "\"}"));
          }
```
and feedback your result.

Upgrading the library to 2.2.0 adds 6k to the code while not bringing any new features.

stylab, 2017-10-13:
Thanks for the fast replay ill test it out tonight when i am at home and report back

stylab, 2017-10-24:
Hi

Sorry for the long test time, but life got in the way
I did finaly find time to test it out
For the panasonic protocol to work we need to define a address

user_config.h

```
#define USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+3k code, 0.3k mem)
#ifdef USE_IR_REMOTE
  #define PanasonicAddress      0x4004           // Panasonic address (Pre data)
#endif
```

and in xdrv_ir_send
```
else if (!strcasecmp_P(protocol, PSTR("PANASONIC")))
  irsend->sendPanasonic(PanasonicAddress, data);
else {
```

ugly as hell but it is working :)
ill do some more testing on my setup but for now with that hack it works
the command still expect a bit value, but we dont use it

arendst, 2017-10-26:
Ok, In version 5.8.0p I've swapped the data and the bits fields for Panasonic and increased the sizie of bits from 8 to 32 bits. This way you should be able to use the Panasonic protocol as follows:
```
IrSend {"Protocol":"Panasonic", "Bits":16388, "Data":<Panasonic data>}
```
Where 16388 is hexadecimal 0x4004.

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst