arendst/Tasmota #1015: Sonoff basic... sensor TEMT6000

dawsmac, 2017-10-12:
Before I spend time looking to see if I can code the TEMT 6000 to work with my Sonoff Basic. Can someone tell me if this is possible etc..

arendst, 2017-10-12:
Why code? It's just a transistor that when connected to the analog input should return a voltage related to illuminance.

dawsmac, 2017-10-12:
What pin would I need to have this connected to ?

davidelang, 2017-10-12:
There's only one analog input on the esp8266, look for the ADC pin

dawsmac, 2017-10-13:
Thanks for the input guys. following this https://iotbytes.wordpress.com/esp8266-pinouts/ I have located pin 6 on the chip and solder a wire I don't see anything in the console?

arendst, 2017-10-13:
Well done. Now to get ADC working you'll have to do some more steps:
1) recompile the program after you enable ADC input in file ``user_config.h`` by adding two slashes in front of line 159:
```
//#define USE_ADC_VCC
```
2) Once the program is loaded you need to select another module as the ``Sonoff Basic`` does not support ADC input. Open de confiuration web page and select module ``Wemos D1 mini``. This enables ADC by default. You will have to add all standard Sonoff Basic IO as defined in file ``sonoff_template.h`` near line 240/247.

dawsmac, 2017-10-13:
Thanks, Arendst... my outside light just became at smarter unit ;)

closed 2017-10-13 by dawsmac

nightcat91, 2017-11-25:
Hello @dawsmac ,

could you help me with something?
I would also like to use this sensor to get a reading about the light situation.
Following my own tries a few month back and reading here I was able to get the analog reading to show up on the webpage.

Did you manage to "convert" this analog value into something useful? like a lux value?