arendst/Tasmota #1016: Module 3 not working with ws2812

MatzeJoerling, 2017-10-13:
Hi,

Sonoff SV flashed with 5.8.0 , Relais working as excepted.
When configuring a WS2812 to any of the outputs (4,5,14) the module reboots and the relais is not working anymore. the WS2812 are working, the LED on board shows that the relais should work but it doesnt "click". no matter if the WS2812 are connected or not. 
reverting back to no WS2812, everything fine, relais is switching again.

regards Matze

arendst, 2017-10-13:
Tested with version 5.8.0k on a wemos d1 mini and works just fine.

[image]

I will dig up a Sonoff SV and try on that one too...

MatzeJoerling, 2017-10-13:
Hi,

perhaps a video shows more than 1000 words, video with everything working:
plain config, nothing on the gpios.
https://youtu.be/77PyjLJAL-Y

then with configured ws2812 on gpio4:
https://youtu.be/GLUGZoh08-k
Listen to the click from the relais, nothing... the other click is my mouse...
also the red led is missing...

both with pulsetimer1 set to 10, i also reset the configuration and started with an empty module.

regards Martin

joecotton, 2017-10-13:
Just to add that I also experienced this behavior, though I haven't spoken up because I haven't had time to narrow down the specifics to make a good report. Reverting to 5.7.x, no problems.

arendst, 2017-10-13:
Confirmed! Will be solved in next (pre)release.

MatzeJoerling, 2017-10-15:
can confirm that it is working now.
will remove the youtube videos.

thanks for the fast response.

regards Matze

closed 2017-10-19 by arendst