arendst/Tasmota #1017: Is it possible to add a ButtonMode command, same as SwitchMode?

calcipher49, 2017-10-13:
For now, it is possible to "disconnect" buttons from relays by defining a specific topic with ButtonTopic command.
However, the only message send to MQTT is 'TOGGLE' when button is pressed or released, similar to SwitchMode=0 for switches.
But for switches, we can change SwitchMode to 1 or 2, and then MQTT messages distinguish states ON and OFF.
It would be interesting to be able to do the same with the buttons, differentiate between states push and released.

Thanks for your great job...

davidelang, 2017-10-13:
switchtopic is what makes switches send messages instead of changing relays

buttontopic does the same thing for buttons.

switchmode changes what messages are generated when the switch opens and closes, 
there is no equivalent for buttons.

calcipher49, 2017-10-13:
You're right,
My question is precisely if it is possible to add the equivalent of switchMode for buttons.

ionciubotaru, 2017-10-14:
Define a switch on the button pin and enjoy switchtopic

calcipher49, 2017-10-14:
> Define a switch on the button pin and enjoy switchtopic
Sorry, do you explain how this can be done?

ionciubotaru, 2017-10-14:
Read this: https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

calcipher49, 2017-10-14:
Ok, thanks very much.
Replace GPIO_KEYx by GPIO_SWTx or GPIO_USER in sonoff_template.h should do the trick...

closed 2017-10-14 by calcipher49

sfanicola, 2018-01-22:
Sorry for asking on a closed issue, but could you explain me how to do the modification?
I'd like to set the GPIO0 button acting like a switch.
I tried to replace GPIO_KEY1 with GPIO_SWT1 but it doesn't work