arendst/Tasmota #1019: adding reverse, clock hand width an color to the clock scheme?

jhlweb, 2017-10-14:
Is it possible to add a reverse/flipped to the clocked scheme and or change the width and color of the hand?
I have a Ikea clock that is made of milky plastic and I need to reverse/flipped the LED sequence because its the only way to mount the LED to the clock facing the wall and it gives a more ambylight effect to the clock.  And I want to be able to change the hands  width and color with MQTT because is meant to go in the child's room as a stay in bed / free to get out and i want to change to color as commands from the server.

I got it working for the clock with this code.. but maybe there is a better way? And I'm not sure where to push the variables to the sysCfg..

`struct RgbColor clr_merge_dim(struct RgbColor c, struct RgbColor  m, uint8_t dim, uint8_t offset)
{ 
  RgbColor ret {0,0,0};

  offset = pow(2,offset);
  #define cmin(a,b) ((a)<(b)?(a):(b))
  ret.R = cmin(c.R + ((m.R / dim)/offset), 255);
  ret.G = cmin(c.G + ((m.G / dim)/offset), 255);
  ret.B = cmin(c.B + ((m.B / dim)/offset), 255);
  return ret;
}

void ws2812_clock()
{
 RgbColor c;
  uint8_t clk_hand_sec_width = 1; //push to sysCfg
  uint8_t clk_hand_min_width = 3; //push to sysCfg
  uint8_t clk_hand_hour_width = 5; //push to sysCfg
  RgbColor clk_hand_sec_clr (255,0,255); //push to sysCfg
  RgbColor clk_hand_min_clr (0,255,0); //push to sysCfg
  RgbColor clk_hand_hour_clr (255,0,0); //push to sysCfg
  boolean clk_reverse = true; //push to sysCfg
  
  strip->ClearTo(0);   // Reset strip
  float newDim = 100 / (float)sysCfg.led_dimmer[0];

  int j = sysCfg.led_pixels;
  int clksize = 600 / j;
  int i = (rtcTime.Second * 10) / clksize;

  if(clk_reverse){ i = j- i; }
  c = strip->GetPixelColor(mod(i, j)); c=clr_merge_dim(c,clk_hand_sec_clr, newDim, 0); strip->SetPixelColor(mod(i, j), c);
  for (uint8_t h =  1; h <= ((clk_hand_sec_width - 1)/2) ; h++) {
    c = strip->GetPixelColor(mod(i-h, j)); c=clr_merge_dim(c,clk_hand_sec_clr, newDim, h); strip->SetPixelColor(mod(i-h, j), c);
    c = strip->GetPixelColor(mod(i+h, j)); c=clr_merge_dim(c,clk_hand_sec_clr, newDim, h); strip->SetPixelColor(mod(i+h, j), c);
  }
  
  i = (rtcTime.Minute * 10) / clksize;
  if(clk_reverse){ i = j- i; }
  c = strip->GetPixelColor(mod(i, j)); c=clr_merge_dim(c,clk_hand_min_clr, newDim, 0); strip->SetPixelColor(mod(i, j), c);
  for (uint8_t h =  1; h <= ((clk_hand_min_width - 1)/2) ; h++) {
    c = strip->GetPixelColor(mod(i-h, j));  c=clr_merge_dim(c,clk_hand_min_clr, newDim, h); strip->SetPixelColor(mod(i-h, j), c);
    c = strip->GetPixelColor(mod(i+h, j));  c=clr_merge_dim(c,clk_hand_min_clr, newDim, h); strip->SetPixelColor(mod(i+h, j), c);
  }

  i = (rtcTime.Hour % 12) * (50 / clksize);
  if(clk_reverse){ i = j- i; }
  c = strip->GetPixelColor(mod(i, j)); c=clr_merge_dim(c,clk_hand_hour_clr, newDim, 0); strip->SetPixelColor(mod(i, j), c);
  for (uint8_t h =  1; h <= ((clk_hand_hour_width - 1)/2) ; h++) {
    c = strip->GetPixelColor(mod(i-h, j));  c=clr_merge_dim(c,clk_hand_hour_clr, newDim, h); strip->SetPixelColor(mod(i-h, j), c);
    c = strip->GetPixelColor(mod(i+h, j));  c=clr_merge_dim(c,clk_hand_hour_clr, newDim, h); strip->SetPixelColor(mod(i+h, j), c);
  } 
  
  ws2812_stripShow();
}`

arendst, 2017-10-18:
Working on it.

jhlweb, 2017-10-18:
@arendst  Great thx.. 

I tried u update some so it can be updated by cmnd but i'm stuck at the LED color in the settings.h and the xdrv_snfled.ino
This is what is working for the width.

for the translation file(s)
```
#define D_CMD_CLOCKREVERSE "ClockReverse"
#define D_CMND_CLOCKSECONDWIDTH "ClocSecondkWidth"
#define D_CMND_CLOCKMINUTEWIDTH "ClockMinuteWidt"
#define D_CMND_CLOCKHOURWIDTH "ClockhHourWidth"
#define D_CMND_CLOCKSECONDCOLOR "ClockSecondColor"
#define D_CMND_CLOCKMINUTECOLOR "ClockMinuteColor"
#define D_CMND_CLOCKHOURCOLOR "ClockHourColor"
```
for Settings.h 
```
  byte          free_475[2];               // 475 

  byte          clock_reverse               // 477
  uint8_t       clock_second_width;         // 478 
  uint8_t       clock_minute_width;         // 479 
  uint8_t       clock_hour_width;           // 47A 
  uint8_t       clock_second_color[3];      // 47B 
  uint8_t       clock_minute_color[3];      // 47E 
  uint8_t       clock_hour_color[3];        // 481 
```

I know i have to do something in settings.h with filling the flash and xdrv_snfled.ino but what i'm clueless for setting the color

```
    if (Settings.version < 0x0508000E) {
       Settings.clock_reverse = CLOCK_REVERSE;
       Settings.clock_second_width = CLOCK_SECOND_WIDTH;
       Settings.clock_minute_width = CLOCK_MINUTE_WIDTH;
       Settings.clock_hour_width = CLOCK_HOUR_WIDTH;
       //Settings.clock_second_color[0] = CLOCK_SECOND_WIDTH; //(255, 0, 255)
       //Settings.clock_minute_color[0] = CLOCK_MINUTE_WIDTH; 
       //Settings.clock_hour_color[0] = CLOCK_HOUR_WIDTH;
    }
```
```
   else if ((11 == light_type) && !strcasecmp_P(type, PSTR(D_CMD_CLOCKREVERSE))) {
    if ((payload >= 0) && (payload <= Settings.led_pixels)) {
      Settings.clock_reverse = payload;
    }
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMD_CLOCKREVERSE "\":%d}"), Settings.clock_reverse);
  }
  else if ((11 == light_type) && !strcasecmp_P(type, PSTR(D_CMND_CLOCKSECONDWIDTH))) {
    if ((payload >= 0) && (payload <= Settings.led_pixels)) {
      Settings.clock_second_width = payload;
    }
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_CLOCKSECONDWIDTH "\":%d}"), Settings.clock_second_width);
  }
  else if ((11 == light_type) && !strcasecmp_P(type, PSTR(D_CMND_CLOCKMINUTEWIDTH))) {
    if ((payload >= 0) && (payload <= Settings.led_pixels)) {
      Settings.clock_minute_width = payload;
    }
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_CLOCKMINUTEWIDTH "\":%d}"), Settings.clock_minute_width);
  }
  else if ((11 == light_type) && !strcasecmp_P(type, PSTR(D_CMND_CLOCKHOURWIDTH))) {
    if ((payload >= 0) && (payload <= Settings.led_pixels)) {
      Settings.clock_hour_width = payload;
    }
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_CLOCKHOURWIDTH "\":%d}"), Settings.clock_hour_width);
  }
```

and changing xdrv_ws2812
```
void Ws2812Clock()
{
  RgbColor c;

  RgbColor hand_second_color CLOCK_SECOND_COLOR; // clock_second_color[0], clock_second_color[1], clock_second_color[2];  //RGB
  RgbColor hand_minute_color CLOCK_MINUTE_COLOR; //clock_minute_color[0], clock_minute_color[1], clock_minute_color[2]; //RGB
  RgbColor hand_hour_color CLOCK_HOUR_COLOR;   //clock_hour_color[0], clock_hour_color[1], clock_hour_color[2];  //RGB

  strip->ClearTo(0); // Reset strip
  int clksize = 600 / (int)Settings.led_pixels;
  Ws2812UpdateHand((RtcTime.second * 10) / clksize, Settings.clock_second_width, hand_second_color);
  Ws2812UpdateHand((RtcTime.minute * 10) / clksize, Settings.clock_minute_width, hand_minute_color);
  Ws2812UpdateHand((RtcTime.hour % 12) * (50 / clksize), Settings.clock_hour_width, hand_hour_color);
```

arendst, 2017-10-20:
Don't bother.

To keep the code footprint as small as possible I have it almost ready using commands Color2, 3 and 4 and Width2, 3 and 4 and SetOption16.

jhlweb, 2017-10-22:
oke.. greet!
can't wait. I have the clocks ready for the firmware.

jhlweb, 2017-10-23:
@arendst  great thx. 

Setoptions16 = Clock Reverse works fine
Width2, 3, 4 and Color2 are also working 
But Color3 and 4 are not working.. the color will not update.  

----- 

and in settings.h 
line 165:   `uint8_t       ws_color[4][3];            // 475`
this first array is length 4 but in drv_snfled.ino
line 647:  ` if ((light_subtype > 1) && !strcasecmp_P(type, PSTR(D_CMND_COLOR)) && (index > 0) && (index <= 5)) {`
it can have an index  of 5 will this not give memory issues?

arendst, 2017-10-24:
You're right regarding color3 and 4. I will investigate.

The index is not a problem as a few lines lower I decrement by two so only array index 0 - 3 is used.

arendst, 2017-10-24:
Pls try again.

jhlweb, 2017-10-25:
Yes this works.

closed 2017-10-25 by arendst

jhlweb, 2017-10-27:
@arendst, Is there any reason why the clock hour hand is not following the minutes? 

`  Ws2812UpdateHand(((RtcTime.hour % 12) * (50 / clksize))+ ((RtcTime.minute * 10) / (12 * clksize)) , WS_HOUR);
`