arendst/Tasmota #1021: Relay is clicking from time to time

eLement87, 2017-10-15:
Hi,

i have a SonOff Pow with Tasmota 5.5.1. (ok... i just saw that the newest Version is 5.8.0l).

So, the SonOff Pow is used to Switch on a Lamp... mostly in the evening. And between this time you here the Relay is clicking 2 - 3 times a day. But it isn't changing the state.  Especially when the Lamp is turned on, the Relay is clicking. 

I searched for this strange behavior, but i can't find some hints in my Logs. The SonOff isn't loss the WiFi Signal, no Reboots, ... i don't know.

A look in the logs, it only prints:

> 15:02:00 MQTT: tele/sonoffLampe/UPTIME = {"Time":"2017-10-15T15:02:00", "Uptime":138}
> 15:03:11 MQTT: tele/sonoffLampe/STATE = {"Time":"2017-10-15T15:03:11", "Uptime":138, "Vcc":3.212, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"iot17", "RSSI":70, "APMac":"26:xx:xx:xx:xx:xx"}}

Sorry if a similar Issue is already discussed. Probably ich searched with wrong words...

Thanks and Best Regards
eLement87

davidelang, 2017-10-15:
increase the logging detail, that will show what's happening

eLement87, 2017-10-15:
Well. I have set the logging to 4 and monitored for around 3 hours... the only output:
Every 20 Seconds the Wifi check, nothing else.

> 20:17:09 Wifi: Checking connection...
> 20:17:09 Wifi: Connected
> 20:17:29 Wifi: Checking connection...
> 20:17:29 Wifi: Connected
> 20:17:39 MQTT: tele/sonoffLampe/STATE = {"Time":"2017-10-15T20:17:39", "Uptime":143, "Vcc":3.222, "POWER":"ON", "Wifi":{"AP":1, "SSID":"iot17", "RSSI":78, "APMac":"26:xx:xx:xx:xx:xx"}}
> 20:17:49 Wifi: Checking connection...
> 20:17:49 Wifi: Connected
> 20:18:10 Wifi: Checking connection...
> 20:18:10 Wifi: Connected
> 20:18:30 Wifi: Checking connection...

Nothing appears when the Relay is randomly clicking.

eLement87, 2017-10-22:
Hi,

sorry for the dalay.

I've analyzed my AP- and Router-Logs if the SonOff diconnects or somthing else (like DHCP Lease or WiFi disconnects). Certently nothing to see. The SonOff is the Client with the longest connection time. 

Maybe a Montagsmodell? ;)

I will try another SonOff Pow and a normal SonOff.

eLement87, 2017-11-01:
The clicking noises comes from another device. My Samsung TV is the culprit. (SonOff is mounted behind the tv...) :sweat_smile:

The Issue can be closed. 
There is no problem with the SonOff neither Tasmota. :wink:

closed 2017-11-01 by eLement87