arendst/Tasmota #1026: Sonfoff basic can't connect to MQTT broker as configured.

internetcontact, 2017-10-16:
I have MQTT broker running on Raspberry pi(static ip) and sonoff is configured to use this. All was working fine till I got new Roomba vacuum and setup on same wifi. Now sonoff is trying to connect to roomba instead of my broker on raspberry.

Sono
[image]
[image]
[image]

ionciubotaru, 2017-10-17:
on serial interface send this command `MqttHost 192.168.1.100`

internetcontact, 2017-10-17:
I have already tried that. Also tried to flash a new sonoff, but same result. The things I don't understand is 
1. "MQTT service found on Roomba-3144460062428500, IP Address 192.168.1.11, Port 1883" - Rommba's ip address is not configured anywhere in sonoff still it finds that service and trying to connect to it.
2. even if I change the config for Mqtt host it don't use it and revert it back to roomba.

Thanks for helping.

ionciubotaru, 2017-10-17:
add this line to user_config_override.h

`#undef USE_DISCOVERY                        // Disable Discovery services for both MQTT and web server`

ionciubotaru, 2017-10-17:
and also send `reset 1` via serial or mqtt

internetcontact, 2017-10-17:
Great advice. This is exactly what I was looking for. Thanks for quick response, saved me lot of time and headache.

I just disabled the MQTT_HOST_DISCOVERY.

closed 2017-10-17 by internetcontact

ionciubotaru, 2017-10-18:
@arendst please consider that MQTT DISCOVER should work only when MQTT HOST is empty

arendst, 2017-10-18:
@ionciubotaru will be in next (pre)release

Bedzaa, 2020-01-12:
> add this line to user_config_override.h
> 
> `#undef USE_DISCOVERY // Disable Discovery services for both MQTT and web server`

Hi, this is link to the same problem I have with Roomba. Could You explain a bit more "how to do that"? I am not that professional and I have no idea where can I find that file. Thanks a lot, Bedza