arendst/Tasmota #1027: Can't flash firmware on Sonoff Dual ver 2.0

ribesx, 2017-10-16:
I tried to flash the firmware on my Sonoff Dual Version 2.0, using a FT232RL FTDI USB adapter, but I can't get it working. I checked the wirings multiple times, connecting TX to TX and RX to RX, and I tried to swap them as well. I'm connecting GPIO0 to the GND for 1/2 seconds while I connect the adapter to the USB port on my PC, but it never works. I tried flashing with Arduino IDE and PlatformIO, and I always get

> `error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

edit21, 2017-10-16:
You need to go tx-rx rx-tx and I always tend to leave gpio0 shorted until I'm done.
Sounds silly but you may want to check which comm port you have configured too.
I have flashed a good few ver2.0 in the last few days - not had an issue yet :(

ribesx, 2017-10-16:
I tried to do that, but it didn't work. I tried installing PlatformIO on Ubuntu and did the procedure but I still can't flash it. I think I'm not shorting gpio0 maybe, I'm using a jumper to connect gpoi0 to the ground, as in the picture
[image]

andyfitter, 2017-10-16:
I managed to grout GPIO via the thin trace on the other side of the board. Maybe you'll have more success there?

ribesx, 2017-10-16:
I already tried that, I also scratched the trace a little bit in order to make a better connection, but no luck yet

andyfitter, 2017-10-16:
When I achieved it via the top track I grounded it from before, to about a second after power was applied. I believe you can tell it has worked when the LED doesn't flash at all. If its flashing in any pattern I don't think you've taken it into Flash mode.

ribesx, 2017-10-17:
So I guess that’s my problem, when I connect the adapter to the USB port there’s always the blue light flashing on the sonoff dual

Norien, 2017-10-18:
Sonoff dual was a huge pain for me. Must have had 20 bad flashes and was getting real frustrated. Tried a new ftdi and it worked first try...

ribesx, 2017-10-18:
I ended up burning my sonoff, I ordered a new one, and I’ll try another ftdi as well.. It’s really frustrating

tyriis, 2017-12-02:
I hold the jumper against the gpio pin from the esp8285 Chip, it finally worked after 20 failed tries with the position shown.

b-runden, 2017-12-18:
Hi all, 
I've also spent days on that "§$%ing sonoff dual v2.0. It wont start flashing mode without having GPIO0 -> GND. 
Neither the button nor the PINS on the board for for GPIO 0 are working in the beginning. (Needs programming first, I guess. It seems to be a specialty for the "ESP8266ex" (likely not the 'pure' 8266...). 
Like @tyriis I connected GND (of one GND pin on the main side) to the resistor's end on the pic of @ribesx ' post. 
that finally worked. (no scratching needed, just pressing a jumper pin to the resistor. (whole time of flashing. Otherwise flashing may stop/break.)

I also tried sonota, however that didn't work either, maybe because I did already the APP Provisioning with iTead... Didn't spend any resource anymore. Flashing with usb serial is pretty fine. 

BTW: Makes no difference if provisioned with iTead's App or not...

calcipher49, 2018-01-12:
I tried to flash my first Sonoff Dual R2 today. But seems impossible to start it into flash mode. 2 hours of multiple failures...
Solved using Sonota and OTA flashing, works like a charm.