arendst/Tasmota #1030: External LED as ON/OFF State

danielpondruff, 2017-10-17:
Hi,

i have a problem.
I would love to see Tasmota in my Coffee Machine.

My Plan is to use es Weemos(or similar device) to Control my Machine. I plan to use a Transistor as a Relay Switch to tap the native Push Button of my Coffee Machine. This is a easy Task.
But i would use the LED of the Coffee Machine as a ON/OFF State.

How could i get the State of the LED if is it on or off? Is this task doable with Tasmota or any other solution ?

Greetings Daniel

ionciubotaru, 2017-10-17:
define a spare pin as switch1 and connect it to machine's led (better use a opto-isolator https://en.wikipedia.org/wiki/Opto-isolator) 

check the wiki for switchtopic and switchmode to configure your switch

danielpondruff, 2017-10-17:
sorry can yo explain this a little bit more ?

ionciubotaru, 2017-10-17:
On hardware side please read https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow#debugging

On software side run the following on your serial interface:
```
switchtopic abc
switchmode1 1
```
then check the mqtt messages Tasmota send when the switch1 goes ON and OFF

closed 2018-05-07 by stale[bot]