arendst/Tasmota #1032: Sonoff POW telemetry update

mkh595, 2017-10-17:
When relay change state On/Off, on web interface are measured values Voltage, Power, Current,... updated immediately. But MQTT message is send only when telemetry period is reached. This leads to discrepancy in HA, where switch is updated immediatly e.g. to ON but Voltage, Power, Current shows still 0.

Can be sent after change of relay state also telemetry data to have correct values and not wait till telemetry period is reach?

`tele/home214/ENERGY = {"Time":"2017-10-17T16:21:02", "Total":74.901, "Yesterday":0.587, "Today":1.297, "Period":1, "Power":79, "Factor":0.91, "Voltage":224, "Current":0.384}`

davidelang, 2017-10-17:
It takes time to measure the power and current values, they are not available 
immediately after the device is turned on. It can take up to 4 seconds to gather 
the values.

calcipher49, 2017-10-17:
if you do not want to wait telemetry period, you can query Energy infos using cmd:
STATUS 8

stefanbode, 2017-10-28:
I do understand the problem and normally would also vote for a telemetrie update, if the power gets changed that is read for example on a per second intervall. But, after doing some investigation there are some not easy to overcome problems. 1. do you really want to flood your MQTT server every second, if there s a change. 2. what is a ‚change‘? More than 1% or 5% from actual value.

All not very easy to Programm on a general way. My workaround is to use ‚teleperiod 15‘. This works OOTB and the delay is maximum 14 seconds.

davidelang, 2017-10-28:
actually, it can take several seconds to calculate the power usage on the POW.

stefanbode, 2017-10-29:
What about the value you see in the web interface. These look to me like calculated on a per second base.

mkh595, 2017-11-25:
_if you do not want to wait telemetry period, you can query Energy infos using cmd:
STATUS 8_
@calcipher49 : This is not solution. HA can have only one topics subscribtion per field.  But command STATUS8 returns different topic and different structure of JSON, energy is in sub level. I don't understand why same values are returned back in two different topics. 
The other complication is to tell to HA when to issue STATUS8 message.
```
14:45:59 MQT: tele/home212/ENERGY = {"Time":"2017-11-25T14:45:58", "Total":542.524, "Yesterday":4.565, "Today":2.454, "Period":0, "Power":0, "Factor":0.00, "Voltage":0, "Current":0.000}
14:45:59 CMD: cmnd/Status 8
14:45:59 MQT: stat/home212/STATUS8 = {"StatusPWR":{"Total":542.524, "Yesterday":4.565, "Today":2.454, "Power":0, "Factor":0.00, "Voltage":0, "Current":0.000}}
```

mkh595, 2017-11-25:
@stefanbode : Why should be MQTT message sent each second? When relay change state, then counter can be set to teleperiod value. E.g. 
```tele_period  = Settings.tele_period -1```
and next second will be MQTT message sent

If relay is set OFF, power and current became 0 immediately, there is no measure necessary.
If relay is set ON, after two seconds is new value there, tested in web interface.

But this is still better than wait several seconds for correct values till teleperiod 15, 30, 60, ... secons is reached.

closed 2018-05-07 by stale[bot]