arendst/Tasmota #1033: Sonoff Th16 | Reset/Boot loop

tpaulic, 2017-10-17:
Dear All,

I have a similar problem. When I sent my MQTT to 'restart', he entered the boot loop and never came out.

I have just managed to solve the problem that I shut down the MQTT and now the device is working but I need MQTT to connect it with openHAB

Does anyone have an idea of ​​how to fix it?

tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff TH", "Version":"5.8.0", "FallbackTopic":"SonofTH16_01", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}

stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF

tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
tele/sonoff/INFO1 {"Module":"Sonoff TH", "Version":"5.8.0", "FallbackTopic":"SonofTH16_01", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/RESULT {"Restart":"Restarting"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:45:20", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}

ets Jan 8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d

ionciubotaru, 2017-10-18:
It is Expresiff bootloader problem. Disconnect serial and everything will be ok. Use web console on the web interface

tpaulic, 2017-10-18:
I disconnected the serial, sonoff still restarting on boot.

This is the log:

stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic", "Version":"5.8.0", "FallbackTopic":"DVES_073754", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
/ikealampa.esp8266/Uptime/ 7490.00
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7490.00
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic", "Version":"5.8.0", "FallbackTopic":"DVES_073754", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
/ikealampa.esp8266/Uptime/ 7490.00
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7491.00
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic", "Version":"5.8.0", "FallbackTopic":"DVES_073754", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
/ikealampa.esp8266/Uptime/ 7491.00
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7491.00
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic", "Version":"5.8.0", "FallbackTopic":"DVES_073754", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7491.00
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7491.00
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
stat/sonoff/RESULT {"Restart":"Restarting"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic", "Version":"5.8.0", "FallbackTopic":"DVES_073754", "GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin", "Hostname":"sonoff-5972", "IPAddress":"10.11.1.175"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-09-18T16:44:29", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/RESULT {"Restart":"1 to restart"}
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
/pumpazatopluvodu.esp8266/Uptime/ 7491.00
/ikealampa.esp8266/Uptime/ 7491.00
/pumpazatopluvodu.esp8266/Uptime/ 7491.00

lobradov, 2018-01-17:
Can you block this device from accessing MQTT Broker and see if's still rebooting?
(I want to test my theory of retained reset command)

Alternatively, you can clear your retained messages from MQTT. Assuming you are using Mosquitto, you should: 
```
sudo service mosquitto stop
sudo rm /var/lib/mosquitto/mosquitto.db
sudo service mosquitto start
```
Adjust your path to where your .db is.

closed 2018-05-31 by stale[bot]

shmikel, 2019-05-25:
> Can you block this device from accessing MQTT Broker and see if's still rebooting?
> (I want to test my theory of retained reset command)
>
> Alternatively, you can clear your retained messages from [...]

That works! Thanks