arendst/Tasmota #1034: Has something changed at some stage with Sonoff Pow relay switching?

mihalski, 2017-10-18:
I have my Sonoff Pow's set up as energy monitoring only. I disable the switch and leave them as always on.

I've been noticing for a while weird things happening when I upgrade the firmware on my wifi router and I just realised that ALL of the Sonoff Pow's are rebooting!

So my fridge, my servers and a bunch of other things which really shouldn't be having their power cut are just being turned off :/

I know that this didn't use to be the case. Is some setting that needs to be enabled that may have been reset?

arendst, 2017-10-19:
Check your ``wificonfig`` setting and select a non restart option.

vshaev, 2017-10-20:
wificonfig 4
sonoff basic. 
Same problem on two modules with the loss of WIFI and if the relay is in an ON state.

pnuding, 2017-11-16:
Same issue here. For a while (5.8.0 or maybe even before) I've seen some of the lights flicker occasionally. 
What happens is that the Sonoffs do a software restart, often it happens to multiple devices at the same time, which would support the theory that it's caused by Wifi, since there was no activity on their MQTT topics at that moment. 
Affected are several Sonoff Basics, a WeMos D1 with a relay and also my Sonoff LEDs. 
In all cases the restarts only seem to happen while the relay is on, I've not caught a single restart without.

Wificonfig on these devices was set to 2, i'll now try 5. They're all on 5.9.1 right now

pnuding, 2017-11-17:
Confirmed - using Wificonfig 5 instead of 2 has stopped the reboots on all devices. Before they were piling up several hundreds over the last month :/
I dug up a non-updated sonoff basic for comparison. It's on 4.1.1a and has remained stable with the relay on for an entire day despite being set to Wificonfig 2. So it certainly smells like something has changed

rcarmo, 2018-03-17:
Referencing #2162, where this behavior fried a lightbulb.

mihalski, 2018-06-07:
I don't know if it's rebooting but it certainly does keep toggling GPIO5 on my Wemos Mini D1 on which I have a relay that controls my garage door. This is a real problem as when I lose wifi the garage door opens and closes continuously :/

ascillato, 2018-06-13:
Hi,

Could you post the output of your `Status 0` command and also the output of the console when this happens?

Thanks

mihalski, 2018-06-24:
```
22:14:32 CMD: Status 0
22:14:32 MQT: stat/garage/STATUS = {"Status":{"Module":18,"FriendlyName":["Garage"],"Topic":"garage","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:14:32 MQT: stat/garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.198:9999/data/firmwares/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T03:42:18","StartupUTC":"2018-06-24T08:32:14","Sleep":0,"BootCount":600,"SaveCount":5417,"SaveAddress":"F4000"}}
22:14:32 MQT: stat/garage/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:14:32 MQT: stat/garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["DobNet","indebuurt2"],"TelePeriod":60,"SetOption":["55800009","55818000"]}}
22:14:32 MQT: stat/garage/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
22:14:32 MQT: stat/garage/STATUS5 = {"StatusNET":{"Hostname":"garage-3594","IPAddress":"192.168.1.116","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:EF:AE:0A","Webserver":2,"WifiConfig":4}}
22:14:32 MQT: stat/garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.163","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EFAE0A","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:14:32 MQT: stat/garage/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 24 12:14:32 2018","Local":"Sun Jun 24 22:14:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"13:47","Sunset":"05:57"}}
22:14:32 MQT: stat/garage/STATUS10 = {"StatusSNS":{"Time":"2018-06-24T22:14:32","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","BME280":{"Temperature":22.8,"Humidity":37.8,"Pressure":1018.1},"BH1750":{"Illuminance":0},"TempUnit":"C"}}
22:14:32 MQT: stat/garage/STATUS11 = {"StatusSTS":{"Time":"2018-06-24T22:14:32","Uptime":"0T03:42:18","Vcc":2.907,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"DobNet","RSSI":66,"APMac":"80:2A:A8:8A:21:B3"}}}
```
As for the output of the console, it's not on Wifi when this happens so I can't get the output of the console :/

ascillato, 2018-07-23:
Hi, try `wificonfig 5`

closed 2018-08-18 by ascillato2

jimmyhall82, 2018-12-21:
I am having a similar issue. Every time my access point is restarted, the sonoff is triggering the relay on. Below is the log. I have wificonfig set to 5, but still seem to have something happening... any help is appreciated.

Log starts from when wifi is lost:

04:08:00 DNS: Initialized
04:08:00 MQT: Attempting connection...
04:08:00 MQT: Connected
04:08:00 MQT: tele/sonoff/LWT = Online (retained)
04:08:00 MQT: cmnd/sonoff/POWER = 
04:08:00 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
04:08:00 MQT: stat/sonoff/POWER1 = ON
04:08:01 MQT: cmnd/garagestate/POWER2 = ON (retained)
04:08:10 MQT: Attempting connection...
04:08:10 MQT: Connect failed to 10.0.1.145:1883, rc -2. Retry in 10 sec
04:08:14 DNS: Initialized
04:08:24 MQT: Attempting connection...
04:08:24 MQT: Connected
04:08:24 MQT: tele/sonoff/LWT = Online (retained)
04:08:24 MQT: cmnd/sonoff/POWER = 
04:08:24 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
04:08:24 MQT: stat/sonoff/POWER1 = ON

Then i need to run my mqt trigger to reset things:

04:09:05 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
04:09:27 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
04:09:27 MQT: stat/sonoff/POWER1 = ON
04:09:29 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
04:09:29 MQT: stat/sonoff/POWER1 = OFF

Now everything is back to normal. But will repeat each time the wifi signal is lost.

jimmyhall82, 2018-12-21:
@ascillato apologies for repeating topics, but i am struggling to work this out. I am using mosquito mqtt broker with hassio, and i cant find where to add the retain: false and also how to clear my mqtt broker database, any help is appreciated

ascillato, 2018-12-21:
Look into your _configuration.yaml_

Search for all your `retain: true` and change them for `retain: false`

You can check in the issues linked for the mosquitto command to delete the retained messages or just uninstall mosquitto and install it again.

jimmyhall82, 2018-12-21:
@ascillato i have done those steps, but as you can see below, i still get "retained" on a number of log items in the sonoff:

05:28:04 MQT: Connected
05:28:04 MQT: tele/sonoff/LWT = Online (retained)
05:28:04 MQT: cmnd/sonoff/POWER = 
05:28:43 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:28:43 MQT: stat/sonoff/POWER1 = ON
05:28:44 MQT: cmnd/garagestate/POWER2 = ON (retained)
05:28:45 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:28:45 MQT: stat/sonoff/POWER1 = OFF
05:28:48 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:28:48 MQT: stat/sonoff/POWER1 = ON
05:28:50 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:28:50 MQT: stat/sonoff/POWER1 = OFF
05:28:53 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:28:53 MQT: stat/sonoff/POWER1 = ON
05:28:54 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:28:54 MQT: stat/sonoff/POWER1 = OFF
05:28:58 MQT: cmnd/garagestate/POWER2 = OFF (retained)

From my configuration.yaml i have the following:

cover:
  - platform: mqtt
    name: "Sonoff Garage"
    state_topic: "cmnd/garagestate/POWER2"
    command_topic: "cmnd/sonoff/POWER"
    payload_open: "ON"
    payload_close: "ON"
    payload_stop: "ON"
    state_open: "ON"
    state_closed: "OFF"
    retain: false
    optimistic: false

retain is false, but as noted, power2 is showing (retained)

ascillato, 2018-12-21:
In Tasmota console what is the output of `PowerRetain`?

It should be zero by default. If you have a 1, that is your problem. Do a `PowerRetain 0`, restart your sonoff and your hassio.

jimmyhall82, 2018-12-21:
05:49:31 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"garagestate","SwitchMode":[0,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":0}}

this is from Tasmota, PowerRetain is 0.

I have restarted Hassio and sonoff, yet still getting this log:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_D57445, GroupTopic sonoffs) Version 6.3.0.15(4b287ab-sonoff)-2_3_0
00:00:00 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
00:00:00 RSL: stat/sonoff/POWER2 = OFF
00:00:00 WIF: Connecting to AP1 Cameras in mode 11N as sonoff-5189...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-5189.local with IP address 10.0.1.121
05:47:21 MQT: Attempting connection...
05:47:21 MQT: Connected
05:47:21 MQT: tele/sonoff/LWT = Online (retained)
05:47:21 MQT: cmnd/sonoff/POWER = 
05:47:21 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.3.0.15(4b287ab-sonoff)","FallbackTopic":"DVES_D57445","GroupTopic":"sonoffs"}
05:47:21 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5189","IPAddress":"10.0.1.121"}
05:47:22 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
05:47:22 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:47:22 MQT: stat/sonoff/POWER1 = OFF
05:47:22 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
05:47:22 MQT: stat/sonoff/POWER2 = OFF
05:47:29 MQT: tele/sonoff/STATE = {"Time":"2018-12-21T05:47:29","Uptime":"0T00:00:14","Vcc":3.202,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Cameras","BSSId":"B6:84:C6:08:E4:B4","Channel":6,"RSSI":80}}
05:47:29 MQT: tele/sonoff/SENSOR = {"Time":"2018-12-21T05:47:29","Switch2":"OFF"}
05:48:32 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:48:32 MQT: stat/sonoff/POWER1 = ON
05:48:32 MQT: cmnd/garagestate/POWER2 = ON (retained)
05:48:33 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:48:33 MQT: stat/sonoff/POWER1 = OFF
05:48:36 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:48:36 MQT: stat/sonoff/POWER1 = ON
05:48:38 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:48:38 MQT: stat/sonoff/POWER1 = OFF
05:49:04 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
05:49:04 MQT: stat/sonoff/POWER1 = ON
05:49:05 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
05:49:05 MQT: stat/sonoff/POWER1 = OFF
05:49:09 MQT: cmnd/garagestate/POWER2 = OFF (retained)

jimmyhall82, 2018-12-21:
@ascillato ok..  fixed it. Thanks for your help, appreciate you taking the time to help a newbie.

ascillato, 2018-12-21:
how you fixed?

jimmyhall82, 2018-12-21:
Correct, I believe the reinstall after setting PowerRetain 0, and rebooting the sonoff and then hassio is what fixed it. Thanks again

makedir, 2024-02-05:
Same issue here on a Delock 11828 https://templates.blakadder.com/delock_11828.html it came with 7.2.0 I updated it to 13.3.0. The wifi signal of the device is really weak, it just works if it is like 1-2m away from the AP. It randomly reboots if you toggle Power1 1/0. I have not seen a pattern in it, but it seems, it reboots if it has weak or no Wifi signal and you do a Power1 1/0. How to fix this?