arendst/Tasmota #1035: I2C sensor reading unstable since 5.8.0l

pascalsaul, 2017-10-18:
Since I updated last night to 5.8.0l my BME280 connected to a Sonoff Dual is unstable:
[image]

tele/dual_wp/SENSOR {"Time":"2017-10-18T17:22:49", "BME280":{"Temperature":-143.9, "Humidity":0.0, "Pressure":1070.6}, "TempUnit":"C"}

I've also disabled seriallog and weblog in the meantime but still wrong reading or disconnected BME280? Also restarted the device several times now. Something changed in the code which affects the sensor reading?

arendst, 2017-10-18:
What gpios do you use for i2c on a sonoff dual?

pascalsaul, 2017-10-18:
GPIO4 and GPIO14 (soldered directly on the ESP)

arendst, 2017-10-18:
If you go back to the previous version does it work fine?
What version is it?

pascalsaul, 2017-10-18:
Well, I need to recompile the 5.8.0k source because I replaced the .bin with the 5.8.0l ;)

Tried to new 5.8.0m but getting compile errors now:
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_main.o
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino: In function 'void Ws2812UpdateHand(int, uint8_t, RgbColor)':
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:134:27: error: 'CLOCK_REVERSE' was not declared in this scope
boolean clock_reverse = CLOCK_REVERSE;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino: In function 'void Ws2812Clock()':
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:150:31: error: 'CLOCK_SECOND_WIDTH' was not declared in this scope
uint8_t hand_second_width = CLOCK_SECOND_WIDTH;

^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:151:31: error: 'CLOCK_MINUTE_WIDTH' was not declared in this scope
uint8_t hand_minute_width = CLOCK_MINUTE_WIDTH;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:152:29: error: 'CLOCK_HOUR_WIDTH' was not declared in this scope
uint8_t hand_hour_width = CLOCK_HOUR_WIDTH;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:153:30: error: expected initializer before 'CLOCK_SECOND_COLOR'
RgbColor hand_second_color CLOCK_SECOND_COLOR;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:154:30: error: expected initializer before 'CLOCK_MINUTE_COLOR'
RgbColor hand_minute_color CLOCK_MINUTE_COLOR;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:155:28: error: expected initializer before 'CLOCK_HOUR_COLOR'
RgbColor hand_hour_color CLOCK_HOUR_COLOR;
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:159:72: error: 'hand_second_color' was not declared in this scope
Ws2812UpdateHand((RtcTime.second * 10) / clksize, hand_second_width, hand_second_color);
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:160:72: error: 'hand_minute_color' was not declared in this scope
Ws2812UpdateHand((RtcTime.minute * 10) / clksize, hand_minute_width, hand_minute_color);
^
C:/Users/user/Downloads/5.8.0m/Sonoff-Tasmota-development/sonoff/xdrv_ws2812.ino:161:75: error: 'hand_hour_color' was not declared in this scope
Ws2812UpdateHand((RtcTime.hour % 12) * (50 / clksize), hand_hour_width, hand_hour_color);
^
Compiling .pioenvs\sonoff\FrameworkArduino\core_esp8266_noniso.o

arendst, 2017-10-18:
You need to copy all files including user_config.h

pascalsaul, 2017-10-18:
I did offcourse ;) Just as always and cleaned it also...

arendst, 2017-10-18:
You cleaned up too much. The missing defines are in the new user_config.h

pascalsaul, 2017-10-18:
5.8.0m is loaded on the Dual now... let's wait some hours...

pascalsaul, 2017-10-19:
[image]

Line is flat again... Any idea what caused this? Is it because I flashed a new version to it?

closed 2017-10-22 by pascalsaul