arendst/Tasmota #1036: Bug: DS18x20 temperature disappear from telemetry after few hours

calcipher49, 2017-10-18:
I have a Sonoff 4CH with a DS18B20 plugged on GPIO2 (with pullup resistancce). Last Tasmota firmware (5.8.0k)
That works well at first, but after a while (6-8 hours), DS18x20 infos are no longer present in telemetry messages (nor in Status 10).
DS18x20 is back only after a soft reset (via web interface or Restart 1 cmd).
This seems to be a bug. please take it into account.

Thanks

AlexTransit, 2017-10-19:
the temperature measurement disappears if there were problems with the wifi connection.
it's enough to connect to WEB and everything works without restarting.

calcipher49, 2017-10-19:
I did other tests and you're right Alex. Loosing DS18x20 infos is linked with temporary failure of wifi connection.
And a good workaround is restart wifi on sonoff using AP 1 cmd.
Nevertheless, it is a bug as temperature is not linked with wifi, and it would be good to solve this issue.

arendst, 2017-10-19:
Which version of the ds18b20 code are you using? The default one or the one enabled in user_config.h?

calcipher49, 2017-10-19:
I use standard version of Tasmota with:
In platformio.ini: env_default=sonoff
In  user_config.h: define USE_DS18x20 enabled (uncommented)

arendst, 2017-10-19:
Pls try again but leave USE_DS18X20 commented. This way you will be using the inbuilt driver for a single ds18b20.

calcipher49, 2017-10-19:
Done, firmware recompiled and updated.
I monitor for stability over time

AlexTransit, 2017-10-21:
for calcipher49
the problem disappeared when using the "ds18b20" library?

calcipher49, 2017-10-21:
Yes, I confirm.
No more ds18b20 disapearance after 2 days.
Issue only with multiple DS18X20 library.

closed 2018-05-07 by stale[bot]