arendst/Tasmota #1037: 4CH Pro External Switches

daspid92, 2017-10-19:
Hi, I've an sonoff 4ch pro and I'd like to add 4 external switches, I know it's possible to use GPIO2 and GPIO1 and 3 for this, for the fourth I've searched where is possible to connect it, and in the schematic seems that GPIO 16 and maybe also GPIO6/7/8/11 (SPI) are available, is it possible to use them for switches? How can I set them? Thanks

ionciubotaru, 2017-10-20:
First read https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro
To define your own switches via web interface go to CONFIGURATION / CONFIGURE MODULE
To add even more switches read https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

TommySharpNZ, 2017-10-20:
Man I wish all these Sonoff devices had terminals for external buttons.... Even if they were just wired into the existing physical buttons...

ionciubotaru, 2017-10-20:
OK, just modify the template as described here  https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

daspid92, 2017-10-20:
@TommySharpNZ but for what I've read the onboard buttons are different from the external switches... @ionciubotaru So I need to edit the template, add the pull up resistor on those pins and it should work?

calcipher49, 2017-10-21:
You can also use buttons 2, 3 & 4 (GPIO9/10/14 ) as Switches.
Simply replace GPIO_KEYx by GPIO_SWTx in template for Sonoff 4CH Pre o.
Even better is set value to GPIO_USER, so you can then change the GPIO type in Web interface (in Configure Module section).

daspid92, 2017-10-23:
@calcipher49 Thanks, it's working! :)

closed 2017-10-23 by daspid92

Wessix, 2018-02-01:
Sry to open this again, calcipher 49 says the absolute maximum number of switches is 4? 
1,2,3 would be (GPIO9/10/14)  what would be 4?
and how would you deal with the Programming mode for Remote if somone presses a button to fast? With switches not possible, but with buttons it is....

calcipher49, 2018-02-01:
Maximum number 4 is Tasmota software limitation, but you can assign any available GPIO as switch.
On 4ch Pro, GPIO 1/2/3 are by default user defined, so can be easily assigned to switches.
Internal buttons are GPIO 0/9/10/14

radebebek, 2018-02-02:
Can you send a schematic how the physical switches are connected with gpio on board?

pearce1340, 2018-02-02:
I would really appreciate if someone could point out where and how I can connect the pins? I know how to re-programme the firmware, but I'm not too great on the electrical component side. Also if you could let me know if you need to pull these pins down using a resistor as well... I know some have a pull down built in. :-)

TommySharpNZ, 2018-02-10:
@daspid92 I have actually just soldered some screw terminals connected the the push buttons already in the Sonoff 4CH... Works a treat, although I wish I could make it tidier....

kajmaj, 2018-03-03:
Is it possible to connect a sensor to 4ch Pro? Desired sensor requires power i.e. 3,3V, GND or GND (it is fine) but is there any free GPIO on board or it is necessary to change one button to sensor, solder there a terminal to connect sensor and therefore lost possibility to control switches by buttons and by MQTT as well.

pearce1340, 2018-03-03:
Yes it is, you can use one of the free GPIO pins to connect it to. Personally I connected it to the Serial Out pin which is also a GPIO pin exposed in Tasmota.

kajmaj, 2018-03-03:
Thanks. To have the info complete - could you let me know where is this pin located?

unscel, 2018-03-30:
@radebebek Did you happen to find a way to attach on board gpio's to physical wall switches?

TheCellMC, 2018-04-10:
Hey. Would someone mind pointing out what GPIO pins I can use to monitor the state of switches and where they are located on the 4CH Pro.

Hugo477, 2018-04-10:
To connect external sensors you can use RX and TX (GPIO1 and 3, in schematic namend E-RX and E-TX)  located in J1. The same as is used for flashing.
Furthermore you shall disable `SerialLog` via WebInterface.

TommySharpNZ, 2018-04-11:
Hi, is it not recommended to solder some external switches in parallel with the 4 push button switches? I've done it and it seems to work without issue.....

ilgrank, 2018-07-05:
@TommySharpNZ : I got a Pro 4CH R2, and so far, I've been unable to make it work as you say: If I close a connection at the back of a button, the press gets detected only on buttons 1-3, and it somehow crashes the ESP.
The buttons are labeled with a different number each , i suspect these are not simple buttons.. 
can you please post a photo of how you did soder yours?
thanks

GPRHINO, 2018-07-06:
@TommySharpNZ I am interested as well, I have a Pro 4Ch on the way and would like to know if this is possible.

Thanks

speedyrazor, 2019-02-14:
I have the 4CH Pro R2, where button 1 is not an issue to connect external switch to.
I just tested with wiring a switch direct to each button, and although it works, it's very slow to respond to the switch (maybe 700ms or more, not instant like it is when you assign switch1 to RX for example). I see someone above mentioning GPIO_USER (looks interesting), but my I don't quite understand. Could someone elaborate on the best way to add external switches to all 4 buttons of Sonoff 4CH Pro R2 please?