arendst/Tasmota #1042: relais? Wemos d1 Mini

Bluelinux, 2017-10-21:
I want to use two outputs with relays that attract only briefly. I do not want to switch from on to off but only a short pulse give so that they attract briefly when you press button to toggle. how can I rewrite this in the tasmota sketch?

davidelang, 2017-10-21:
look at the pulsetime command

Bluelinux, 2017-10-21:
I've watched and unfortunately I did not understand it. I do not know where I need to change.

davidelang, 2017-10-21:
you issue the command pulsetimeN time

where N is which relay you are setting, and time is the number of tenths of 
secons (up to 11.1 seconds), or N-100 is the number of seconds (12 seconds to 
64800 seconds)

After you have issued the pulsetime command, then you issue an power on command 
as normal and the relay will turn off after the pulsetime setting.

you can issue the pulsetime command via mqtt, or via the web console on the 
device.

closed 2018-05-07 by stale[bot]