arendst/Tasmota #1043: JSON XPath Parse Value

Hollako, 2017-10-21:
Hello All,

Please I need a help to extract the power stat value from the JSON Expression below after i send a request from a web "http://<IP ADDRESS>/cm?cmnd=Power" so i get this message as result:

RESULT = {"POWER":"ON"}
POWER1 = ON

If I try to write POWER in JSONPath Expression to parse the power state value I got an error.

Thank you in advance.

Regards,

ionciubotaru, 2017-10-22:
1. Use MQTT - the messages are already in json format
2. If you don't like MQTT - what interpretor do you use (command.com, bash, python, ruby, other ...) ?

Hollako, 2017-10-22:
Actually i use Openremote and it does not support MQTT, i usually using the HTTP commands with a JSON replies where i parse the values using a simple JSON Expression.

But the problem here is that the word "Result =" before the {...} is my problem. When i delete the "Result =" and try the phrase in an online JSONPath tester it works. but as long as i get the "Result =" before the JSON data starts here is my problem.

So the solution that i need is how to bypass the "Result =" in the JSONPath Expression ?

Thank you,

fr3nsis, 2017-10-29:
webserver.ino around line 1437 replace
`WebServer->send(200, FPSTR(HDR_CTYPE_PLAIN), message);`
with
```
 message = message.substring (message.indexOf("{"),message.lastIndexOf("}")+1);
 WebServer->send(200, FPSTR(HDR_CTYPE_JSON),message);
```

closed 2018-05-07 by stale[bot]