arendst/Tasmota #1044: Translation for Polish language

roblad, 2017-10-21:
I am attaching Polish language pl-PL for adding to source if you could.

pl-PL.h

--------------CODE start----------------
 /*
  pl-PL.h - localization for Polish without fonetick - Poland for Sonoff-Tasmota
  Copyright (C) 2017  Theo Arends (translated by roblad - Robert L.)
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _LANGUAGE_PL_PL_H_
#define _LANGUAGE_PL_PL_H_

/*************************** ATTENTION *******************************\
 *
 * Due to memory constraints only UTF-8 is supported.
 * To save code space keep text as short as possible.
 * Time and Date provided by SDK can not be localized (yet).
 * Use online command StateText to translate ON, OFF, HOLD and TOGGLE.
 * Use online command Prefix to translate cmnd, stat and tele.
 *
\*********************************************************************/

// "2017-03-07T11:08:02" - ISO8601:2004
#define D_YEAR_MONTH_SEPARATOR "-"
#define D_MONTH_DAY_SEPARATOR "-"
#define D_DATE_TIME_SEPARATOR "T"
#define D_HOUR_MINUTE_SEPARATOR ":"
#define D_MINUTE_SECOND_SEPARATOR ":"

#define D_DAY3LIST "PonWtoSroCzwPiaSobNie"
#define D_MONTH3LIST "StyLutMarKwiMajCzeLipSieWrzPazLisGru"

// Non JSON decimal separator
#define D_DECIMAL_SEPARATOR ","

// Common
#define D_ABORTED "Anulowany"
#define D_ACTIVE "Aktywny"
#define D_ADDRESS "Addres"
#define D_ADMIN "Admin"
#define D_AIR_QUALITY "Jakosc powietrza"
#define D_AIRQUALITY "JakPowietrza"
#define D_AP "AP"                    // Access Point
#define D_APMAC_ADDRESS "APMac"
#define D_APPENDED "Dolaczony"
#define D_AS "jak"
#define D_AUTO "AUTO"
#define D_BAUDRATE "Szybkosc transmisji"
#define D_BLINK "Miganie"
#define D_BLINKOFF "MiganieWyl"
#define D_BOOTVERSION "Uruchomienie"
#define D_BOOT_COUNT "Licznik uruchomienia"
#define D_BOOTCOUNT "LiczUruchomienia"
#define D_BRIGHTLIGHT "Jasny"
#define D_BUILDDATETIME "ZbudowanoDataCzas"
#define D_BUTTON "Przycisk"
#define D_CELSIUS "Celciusza"
#define D_CODE "kod"                // Button code
#define D_COLDLIGHT "Zimny"
[... 595 lines cut ...]
#define D_CMND_POWERLOW "PowerLow"
#define D_CMND_POWERHIGH "PowerHigh"
#define D_CMND_VOLTAGELOW "VoltageLow"
#define D_CMND_VOLTAGEHIGH "VoltageHigh"
#define D_CMND_CURRENTLOW "CurrentLow"
#define D_CMND_CURRENTHIGH "CurrentHigh"
#define D_CMND_ENERGYRESET "EnergyReset"
#define D_CMND_HLWPCAL "HlwPcal"
#define D_CMND_HLWPSET "HlwPset"
#define D_CMND_HLWUCAL "HlwUcal"
#define D_CMND_HLWUSET "HlwUset"
#define D_CMND_HLWICAL "HlwIcal"
#define D_CMND_HLWISET "HlwIset"
#define D_CMND_MAXPOWER "MaxPower"
#define D_CMND_MAXPOWERHOLD "MaxPowerHold"
#define D_CMND_MAXPOWERWINDOW "MaxPowerWindow"
#define D_CMND_SAFEPOWER "SafePower"
#define D_CMND_SAFEPOWERHOLD "SafePowerHold"
#define D_CMND_SAFEPOWERWINDOW "SafePowerWindow"
#define D_CMND_MAXENERGY "MaxEnergy"
#define D_CMND_MAXENERGYSTART "MaxEnergyStart"

#endif  // `_LANGUAGE_PL_PL_H_

--------------CODE end----------------

arendst, 2017-10-21:
Thnx! Will be in next (pre)release.

arendst, 2017-10-21:
Pls try to shorten some strings being to long as compilation currently fails.

I suggest you compile your version and solve the strings which are too long. A quick review showed that the strings at line 456 - 474 need to be 8 characters at most.

roblad, 2017-10-21:
I do not know what changes are needed for adding PL language where and which files some changes need to be done.

roblad, 2017-10-21:
I have only

sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]

arendst, 2017-10-21:
If you use platformio then look in the platformio.ini file how the dutch or german version is selected.

If you use arduino ide make change in file user_config.h at the top where you can find the dutch and german defines too. 

About the 8 character see my previous answer about  the offending line numbers.

roblad, 2017-10-21:
I changed betwen 456-574 it not help 

my platformio.ini added

; Sonoff multiple DS18x20-PL
[env:sonoff-ds18x20-PL]
platform = espressif8266
framework = arduino
; board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=707 -DUSE_DS18x20 -DMESSZ=600 -DMY_LANGUAGE=pl-PL
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, OneWire

// -- Localization --------------------------------
//#define MY_LANGUAGE            en-GB           // Enabled by Default
//#define MY_LANGUAGE            nl-NL
//#define MY_LANGUAGE            de-DE
#define MY_LANGUAGE            pl-PL

ERRORS

Project
Sonoff-Tasmota
api
upload-arduino.php
arduino
version 2.3.0
version 2.4.0-rc1
boards.txt
platform.txt
version 2.4.0-rc2
espupload.py
conf
platformio.ini
sonoff_template.h
sonoff.h
user_config_override.h
user_config.h
lib
ArduinoJson
IRremoteESP8266
NeoPixelBus_by_Makuna
OneWire
PubSubClient
readme.txt
pio
sonoff
.gitattributes
.gitignore
.travis.yml
ctags.tmp
esp8266.flash.1m0.ld
firmware.bin
firmware.elf
LICENSE.txt
platformio.ini
README.md

Project
Sonoff-Tasmota
api
upload-arduino.php
arduino
version 2.3.0
version 2.4.0-rc1
boards.txt
platform.txt
version 2.4.0-rc2
espupload.py
conf
platformio.ini
sonoff_template.h
sonoff.h
user_config_override.h
user_config.h
lib
ArduinoJson
IRremoteESP8266
NeoPixelBus_by_Makuna
OneWire
PubSubClient
readme.txt
pio
sonoff
.gitattributes
.gitignore
.travis.yml
ctags.tmp
esp8266.flash.1m0.ld
firmware.bin
firmware.elf
LICENSE.txt
platformio.ini
README.md
103
104
105
106
107
108
; Serial Monitor options
monitor_baud = 115200
 
; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu
 

6.6 s
platformio run
|   |-- <ESP8266WiFi> v1.0
|-- <ArduinoJson> v5.8.3
|-- <Ticker> v1.0
|-- <DNSServer> v1.1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0

|-- <ESP8266WiFi> v1.0
Compiling .pioenvs\sonoff-ds18x20-PL\src\sonoff.ino.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\Stream.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\StreamString.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\Tone.o
In file included from E:/Sonoff-Tasmota/sonoff/sonoff.ino:35:0:
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
};
^
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
sonoff\sonoff_template.h:141:1: error: initializer-string for array of chars is too long [-fpermissive]
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\Updater.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\WMath.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\WString.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\abi.o
E:/Sonoff-Tasmota/sonoff/xdrv_domoticz.ino:57:22: error: initializer-string for array of chars is too long [-fpermissive]
D_DOMOTICZ_CURRENT };
^
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\base64.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\cbuf.o
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\cont.o
*** [.pioenvs\sonoff-ds18x20-PL\src\sonoff.ino.o] Error 1
Compiling .pioenvs\sonoff-ds18x20-PL\FrameworkArduino\cont_util.o
 [ERROR] Took 4.80 seconds
 
 [SUMMARY]
Environment sonoff              [SKIP]
Environment sonoff-NL           [SKIP]
Environment sonoff-DE           [SKIP]
Environment sonoff-minimal      [SKIP]
Environment sonoff-ds18x20      [SKIP]
Environment sonoff-ds18x20-PL   [ERROR]
 [ERROR] Took 4.80 seconds
 
Linter
Severity 	Provider 	Description	Line 
PIO Buildplatformio.ini0001:1
LFUTF-8INIgit+
master565 files19:52

arendst, 2017-10-21:
Ah the ones with a prefix of 1i can only be 6 characters long to make a total of 8 characters

roblad, 2017-10-21:
I have changed as you sugested is better but not well

Project
Sonoff-Tasmota
api
upload-arduino.php
arduino
version 2.3.0
version 2.4.0-rc1
boards.txt
platform.txt
version 2.4.0-rc2
espupload.py
conf
platformio.ini
sonoff_template.h
sonoff.h
user_config_override.h
user_config.h
lib
ArduinoJson
IRremoteESP8266
NeoPixelBus_by_Makuna
OneWire
PubSubClient
readme.txt
pio
sonoff
.gitattributes
.gitignore
.travis.yml
ctags.tmp
esp8266.flash.1m0.ld
firmware.bin
firmware.elf
LICENSE.txt
platformio.ini
README.md
platformio.ini
PlatformIO Home

11.4 s
platformio run
Compiling .pioenvs\sonoff-PL\FrameworkArduino\core_esp8266_wiring_shift.o
Compiling .pioenvs\sonoff-PL\FrameworkArduino\debug.o
Compiling .pioenvs\sonoff-PL\FrameworkArduino\heap.o
E:/Sonoff-Tasmota/sonoff/xdrv_domoticz.ino:57:22: error: initializer-string for array of chars is too long [-fpermissive]
D_DOMOTICZ_CURRENT };
^
Compiling .pioenvs\sonoff-PL\FrameworkArduino\libb64\cdecode.o

Compiling .pioenvs\sonoff-PL\FrameworkArduino\libb64\cencode.o
Compiling .pioenvs\sonoff-PL\FrameworkArduino\libc_replacements.o
*** [.pioenvs\sonoff-PL\src\sonoff.ino.o] Error 1
 [ERROR] Took 6.65 seconds
 
 [SUMMARY]
Environment sonoff              [SKIP]
Environment sonoff-NL           [SKIP]
Environment sonoff-DE           [SKIP]
Environment sonoff-minimal      [SKIP]
Environment sonoff-ds18x20      [SKIP]
Environment sonoff-ds18x20-PL   [SKIP]
Environment sonoff-PL           [ERROR]
 [ERROR] Took 6.65 seconds
 
Linter
Severity 	Provider 	Description	Line 
PIO BuildPlatformIO Home
git+
master565 files20:17

arendst, 2017-10-21:
I think D_DOMOTICZ_TEMP_HUM_BARO is 1 character too long.

roblad, 2017-10-21:
OK, comiled

CODE

/*
  pl-PL.h - localization for Polish without fonetick - Poland for Sonoff-Tasmota
  Copyright (C) 2017  Theo Arends (translated by roblad - Robert L.)
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _LANGUAGE_PL_PL_H_
#define _LANGUAGE_PL_PL_H_

/*************************** ATTENTION *******************************\
 *
 * Due to memory constraints only UTF-8 is supported.
 * To save code space keep text as short as possible.
 * Time and Date provided by SDK can not be localized (yet).
 * Use online command StateText to translate ON, OFF, HOLD and TOGGLE.
 * Use online command Prefix to translate cmnd, stat and tele.
 *
\*********************************************************************/

// "2017-03-07T11:08:02" - ISO8601:2004
#define D_YEAR_MONTH_SEPARATOR "-"
#define D_MONTH_DAY_SEPARATOR "-"
#define D_DATE_TIME_SEPARATOR "T"
#define D_HOUR_MINUTE_SEPARATOR ":"
#define D_MINUTE_SECOND_SEPARATOR ":"

#define D_DAY3LIST "PonWtoSroCzwPiaSobNie"
#define D_MONTH3LIST "StyLutMarKwiMajCzeLipSieWrzPazLisGru"

// Non JSON decimal separator
#define D_DECIMAL_SEPARATOR ","

// Common
#define D_ABORTED "Anulowany"
#define D_ACTIVE "Aktywny"
#define D_ADDRESS "Addres"
#define D_ADMIN "Admin"
#define D_AIR_QUALITY "Jakosc powietrza"
#define D_AIRQUALITY "JakPowietrza"
#define D_AP "AP"                    // Access Point
#define D_APMAC_ADDRESS "APMac"
#define D_APPENDED "Dolaczony"
#define D_AS "jak"
#define D_AUTO "AUTO"
#define D_BAUDRATE "Szybkosc transmisji"
#define D_BLINK "Miganie"
#define D_BLINKOFF "MiganieWyl"
#define D_BOOTVERSION "Uruchomienie"
#define D_BOOT_COUNT "Licznik uruchomienia"
#define D_BOOTCOUNT "LiczUruchomienia"
#define D_BRIGHTLIGHT "Jasny"
#define D_BUILDDATETIME "ZbudowanoDataCzas"
#define D_BUTTON "Przycisk"
#define D_CELSIUS "Celciusza"
#define D_CODE "kod"                // Button code
#define D_COLDLIGHT "Zimny"
#define D_COMMAND "Komenda"
[... 592 lines cut ...]

// Commands xsns_hlw8012.ino
#define D_CMND_POWERLOW "PowerLow"
#define D_CMND_POWERHIGH "PowerHigh"
#define D_CMND_VOLTAGELOW "VoltageLow"
#define D_CMND_VOLTAGEHIGH "VoltageHigh"
#define D_CMND_CURRENTLOW "CurrentLow"
#define D_CMND_CURRENTHIGH "CurrentHigh"
#define D_CMND_ENERGYRESET "EnergyReset"
#define D_CMND_HLWPCAL "HlwPcal"
#define D_CMND_HLWPSET "HlwPset"
#define D_CMND_HLWUCAL "HlwUcal"
#define D_CMND_HLWUSET "HlwUset"
#define D_CMND_HLWICAL "HlwIcal"
#define D_CMND_HLWISET "HlwIset"
#define D_CMND_MAXPOWER "MaxPower"
#define D_CMND_MAXPOWERHOLD "MaxPowerHold"
#define D_CMND_MAXPOWERWINDOW "MaxPowerWindow"
#define D_CMND_SAFEPOWER "SafePower"
#define D_CMND_SAFEPOWERHOLD "SafePowerHold"
#define D_CMND_SAFEPOWERWINDOW "SafePowerWindow"
#define D_CMND_MAXENERGY "MaxEnergy"
#define D_CMND_MAXENERGYSTART "MaxEnergyStart"

#endif  // _LANGUAGE_PL_PL_H_

roblad, 2017-10-21:
my platformio.ini I had to use sonoff-ds18x20-PL as additional section for DS compilation

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
;env_default = sonoff
;env_default = sonoff-NL
;env_default = sonoff-DE
;env_default = sonoff-PL
; env_default = sonoff-minimal
;env_default = sonoff-ds18x20
env_default = sonoff-ds18x20-PL

; *** Sonoff et al
[env:sonoff]
platform = espressif8266
framework = arduino
;board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

; *** Sonoff et al
[env:sonoff-NL]
platform = espressif8266
framework = arduino
;board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512 -DMY_LANGUAGE=nl-NL
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

; *** Sonoff et al
[env:sonoff-DE]
platform = espressif8266
framework = arduino
;board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512 -DMY_LANGUAGE=de-DE
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

; *** Sonoff et al
[env:sonoff-PL]
platform = espressif8266
framework = arduino
;board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512 -DMY_LANGUAGE=pl-PL
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

; Sonoff minimal
[env:sonoff-minimal] ; Placeholder to be configured
platform = espressif8266
framework = arduino
;board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512 -DBE_MINIMAL
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

; Sonoff multiple DS18x20
[env:sonoff-ds18x20]
platform = espressif8266
framework = arduino
; board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=707 -DUSE_DS18x20 -DMESSZ=600
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, OneWire

; Sonoff multiple DS18x20-PL
[env:sonoff-ds18x20-PL]
platform = espressif8266
framework = arduino
; board = esp01_1m
board = esp07
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=707 -DUSE_DS18x20 -DMESSZ=600 -DMY_LANGUAGE=pl-PL
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, OneWire

; Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/http-uploader.py

roblad, 2017-10-21:
Could you add in this release also one more device just after SUPLA1 - it is modyfication of SUPLA1 I had 30 devices and will be more :-)

TASMOTA_RLPB

.....
 { "TasSupla-RLPB",   // RL_PB_TASMOTA hardware (ESP8266)

     GPIO_USER,        // GPIO00 D3 Wemos Button Shield
     0,        // GPIO01 TX Serial RXD
     GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
     0,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 D2 Wemos I2C SDA
     GPIO_REL1,        // GPIO05 D1 Wemos I2C SCL / Wemos Relay Shield (0 = Off, 1 = On) / Wemos WS2812B RGB led Shield
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        // GPIO12 D6
     GPIO_REL2,        // GPIO13 D7
     GPIO_USER,        // GPIO14 D5
     0,        // GPIO15 D8
     GPIO_LED1,        // GPIO16 D0 Wemos Wake
     GPIO_ADC0         // ADC0   A0 Analog input
  },
....

maciejtarmas, 2017-10-21:
Some quick and dirty spelling, grammar, capitalization and misc. corrections. Haven't gone through all of this, mind you.

``#define D_ADDRESS "Addres"``

#define D_ADDRESS "Adres"

``#define D_CELSIUS "Celciusza"``

#define D_CELSIUS "Celsiusza"

``#define D_FAHRENHEIT "Farenheita"``

#define D_FAHRENHEIT "Fahrenheita"

``#define D_NO "No"``

#define D_NO "Nie"

``#define D_POWERFACTOR "Wspolczynik"``

#define D_POWERFACTOR "WspolczynikMocy"

``#define D_RESTART_REASON "Przyczyna Restartu"``

#define D_RESTART_REASON "Przyczyna restartu"

``#define D_RESTARTREASON "RestartReason"``

#define D_RESTARTREASON "PrzyczynaRestartu"

``#define D_RETAINED "zachowane"``

#define D_RETAINED "Zachowane"

``#define D_SELECTED "wybrany"``

#define D_SELECTED "Wybrany"

``#define D_BYTES "Baitow"``

#define D_BYTES "Bajtow"

``#define D_STARTED "Wystartowany"``

#define D_STARTED "Uruchomiony"

``#define D_SUBNET_MASK "Subnet Mask"``

#define D_SUBNET_MASK "Maska podsieci"

``#define D_SUBNETMASK "Subnetmask"``

#define D_SUBNETMASK "MaskaPodsieci"

``#define D_TRANSMIT "Transmituj"``

#define D_TRANSMIT "Wyslij"

``#define D_UPGRADE "aktualizacja"``

#define D_UPGRADE "Aktualizacja"

``#define D_VCC "Vcc"``

#define D_VCC "VCC"

``#define D_USE_DEFAULTS "Uzyj domyslne"``

#define D_USE_DEFAULTS "Uzyj domyslnych"

[... 238 lines cut ...]
#define D_UPLOAD_ERR_3 "Magiczny bajt jest inny niz 0xE9"

``#define D_UPLOAD_ERR_4 "Rozmiar programu jest większy niż rzeczywisty rozmiar pamięci flash"``

#define D_UPLOAD_ERR_4 "Rozmiar programu jest wiekszy niz rzeczywisty rozmiar pamieci flash"

``#define D_UPLOAD_ERR_5 "Wgrywanie, bufor niezgodności stanu porównywanych bitów"``

#define D_UPLOAD_ERR_5 "Wgrywanie, bufor niezgodnosci stanu porownywanych bitow"

``#define D_UPLOAD_ERR_6 "Wgrywanie bledne. Uruchomiono logowanie poziom 3"``

#define D_UPLOAD_ERR_6 "Blad wgrywania. Uruchomiono zapis do dziennika na poziomie 3"

``#define D_UPLOAD_ERROR_CODE "Wgrywanie error code"``

#define D_UPLOAD_ERROR_CODE "Blad wgrywania"

``#define D_ENABLE_WEBLOG_FOR_RESPONSE "Wlacz web logowanie 2, jesli oczekiwana iest odpowiedz"``

#define D_ENABLE_WEBLOG_FOR_RESPONSE "Wlacz poziom 2 zapisu Weblog, jesli oczekiwana jest odpowiedz"

``#define D_NEED_USER_AND_PASSWORD "Wymagany uzytkownik=<username>&password=<password>"''

#define D_NEED_USER_AND_PASSWORD "Wymagany uzytkownik=<username>&haslo=<password>"

roblad, 2017-10-22:
looks better :-)

maciejtarmas, 2017-10-22:
Thanks :)

I'll give it some more polishing (pun intended) some time next week.

roblad, 2017-10-22:
please attach full file after correction Theo will include to next release.

roblad, 2017-10-22:
Ok, I have changed according your correction
LATEST CODE pl-PL.h

```
/*
  pl-PL.h - localization for Polish without fonetick - Poland for Sonoff-Tasmota
  Copyright (C) 2017  Theo Arends (translated by roblad - Robert L.)
  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.
  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.
  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _LANGUAGE_PL_PL_H_
#define _LANGUAGE_PL_PL_H_

/*************************** ATTENTION *******************************\
 *
 * Due to memory constraints only UTF-8 is supported.
 * To save code space keep text as short as possible.
 * Time and Date provided by SDK can not be localized (yet).
 * Use online command StateText to translate ON, OFF, HOLD and TOGGLE.
 * Use online command Prefix to translate cmnd, stat and tele.
 *
\*********************************************************************/

// "2017-03-07T11:08:02" - ISO8601:2004
#define D_YEAR_MONTH_SEPARATOR "-"
#define D_MONTH_DAY_SEPARATOR "-"
#define D_DATE_TIME_SEPARATOR "T"
#define D_HOUR_MINUTE_SEPARATOR ":"
#define D_MINUTE_SECOND_SEPARATOR ":"

#define D_DAY3LIST "PonWtoSroCzwPiaSobNie"
#define D_MONTH3LIST "StyLutMarKwiMajCzeLipSieWrzPazLisGru"

// Non JSON decimal separator
#define D_DECIMAL_SEPARATOR ","

// Common
#define D_ABORTED "Anulowany"
#define D_ACTIVE "Aktywny"
#define D_ADDRESS "Adres"
#define D_ADMIN "Admin"
#define D_AIR_QUALITY "Jakosc powietrza"
#define D_AIRQUALITY "JakPowietrza"
#define D_AP "AP"                    // Access Point
#define D_APMAC_ADDRESS "APMac"
#define D_APPENDED "Dolaczony"
#define D_AS "jak"
#define D_AUTO "AUTO"
#define D_BAUDRATE "Szybkosc transmisji"
#define D_BLINK "Miganie"
#define D_BLINKOFF "MiganieWyl"
#define D_BOOTVERSION "Uruchomienie"
#define D_BOOT_COUNT "Licznik uruchomienia"
#define D_BOOTCOUNT "LiczUruchomienia"
#define D_BRIGHTLIGHT "Jasny"
#define D_BUILDDATETIME "ZbudowanoDataCzas"
#define D_BUTTON "Przycisk"
#define D_CELSIUS "Celsiusza"
#define D_CODE "kod"                // Button code
#define D_COLDLIGHT "Zimny"
#define D_COMMAND "Komenda"
[... 593 lines cut ...]
// Commands xsns_hlw8012.ino
#define D_CMND_POWERLOW "PowerLow"
#define D_CMND_POWERHIGH "PowerHigh"
#define D_CMND_VOLTAGELOW "VoltageLow"
#define D_CMND_VOLTAGEHIGH "VoltageHigh"
#define D_CMND_CURRENTLOW "CurrentLow"
#define D_CMND_CURRENTHIGH "CurrentHigh"
#define D_CMND_ENERGYRESET "EnergyReset"
#define D_CMND_HLWPCAL "HlwPcal"
#define D_CMND_HLWPSET "HlwPset"
#define D_CMND_HLWUCAL "HlwUcal"
#define D_CMND_HLWUSET "HlwUset"
#define D_CMND_HLWICAL "HlwIcal"
#define D_CMND_HLWISET "HlwIset"
#define D_CMND_MAXPOWER "MaxPower"
#define D_CMND_MAXPOWERHOLD "MaxPowerHold"
#define D_CMND_MAXPOWERWINDOW "MaxPowerWindow"
#define D_CMND_SAFEPOWER "SafePower"
#define D_CMND_SAFEPOWERHOLD "SafePowerHold"
#define D_CMND_SAFEPOWERWINDOW "SafePowerWindow"
#define D_CMND_MAXENERGY "MaxEnergy"
#define D_CMND_MAXENERGYSTART "MaxEnergyStart"

#endif  // _LANGUAGE_PL_PL_H_
```

maciejtarmas, 2017-10-22:
@roblad 

I have forked the repo at https://github.com/maciejtarmas/Sonoff-Tasmota and added the translation file. You can edit it there. Once it's finished, we can make a pull request.

roblad, 2017-10-22:
I forked and changed, additionaly I send pool requwes for my changes

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst