arendst/Tasmota #1046: internal LED behaviour programmable

Andirs55, 2017-10-22:
Hello it would be nice to have a programmable internal LED behaviour.
It would be especially nice if one could switch off the blinking when the wifi connection gets lost. 
This is particularly annoying at night in the bedroom.

Thanks

pascalsaul, 2017-10-22:
I must admit... my girlfriend aks the same last night... can you put the blue light off please :)
👍

jetema, 2017-10-23:
If you are talking about sonoff, internal led, it can be switched off easily, have a look at the commands section
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
See LedState and LedPower

Andirs55, 2017-10-23:
Hi jetema,

thanks for the tip. But neither LedState nor LedPower have an effect on the blinking when there is no wifi connection. On my Sonoff Basic it blinks 13 times, then makes a break of ~30s and then blinks 13 times and so on.

I also tried BlinkTime and BlinkCount with no success.

jetema, 2017-10-23:
I also switch off all my wifi by night, I have no blinks at all, so do that ,
ledpower 0, wificonfig 5 ( or wificonfig 4 if you don't have Tasmota 5.8.0 ) and setoption 0 
and let me know your results

Andirs55, 2017-10-24:
Thanks a lot, wificonfig 5 did the trick.

closed 2017-10-24 by Andirs55