arendst/Tasmota #1048: Sonoff T1 3-gang switch (version with wifi and RF): Random switching of 1-st channel.

aksell1981, 2017-10-22:
Before I installed Tasmota on Sonoff basic and installation pass smooth (via serial connection) without any problems.
To installed Tasmota on my Sonoff T1 switch I used “SonOTA Espressif2Arduino Tasmota without compiling” which described in WIKI. OTA process of installation failed somewhere in the middle of process and I got bricked device.
When nothing to lose I tried “serial” upload and installed tasmota firmware successfully and manage to bind the switch with Openhab.
But 1st channel of the switch toggles every few minutes with no reason. The rest works OK. Here is a log when switch triggering byitself :
17:01:19 WIF: Checking connection...
17:01:19 WIF: Connected
**17:01:23 APP: Button 1 multi-press 1**
17:01:24 MQT: stat/sonoff_3t/RESULT = {"POWER1":"ON"}
17:01:24 MQT: stat/sonoff_3t/POWER1 = ON
17:01:25 CFG: Saved to flash at FB, Count 473, Bytes 1648
17:17:10 CFG: Saved to flash at F9, Count 483, Bytes 1648
**17:17:10 APP: Button 1 multi-press 1**
17:17:11 MQT: stat/sonoff_3t/RESULT = {"POWER1":"OFF"}
17:17:11 MQT: stat/sonoff_3t/POWER1 = OFF
17:17:11 CFG: Saved to flash at F8, Count 484, Bytes 1648
17:17:20 WIF: Checking connection...
17:17:20 WIF: Connected

Any help? Thanks in advance.

aksell1981, 2017-10-29:
Ok...i solved by my own. There was nothing wrong with software.

closed 2017-10-29 by aksell1981

kirkyb, 2018-01-06:
Hi. Please can you explain how you managed to flash the T1 using serial connection?

aksell1981, 2018-01-11:
I used similar procedure as for Sonoff Touch.  
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Touch

FatBeard, 2018-03-04:
Hi akshell. I’m having exact same issue. How did you resolve it?

aksell1981, 2018-03-04:
Hello. I just cut conduction path between two pins as on picture below. Chip from left side is responsible for sending commands in case of activation by RF. It seems 1st channel was triggering randomly  by RF, not WIFI commands. At least for me after this modification this problem solved.
[image]

FatBeard, 2018-03-04:
Hey aksell, that looks exactly like my issue. I’ll give it a try. Did you figure this out yourself or is there more documentation on it somewhere. I’d love to know what the chip on the right hand side is. I can’t find any information on it. I’d also love to know why this only became a problem after I programmed tasmota. I’m not blaming software but either my handling of the device or the programming must have done something. For instance, did holding down button 1 to go into program mode cause it or was it an intrinsic fault in this version of the t1. I notice our version has a different synoxo chip (8 leg) than that in wiki.

FatBeard, 2018-03-04:
Actually forgot to say thanks for getting back to me. 

Another theory is maybe I put the device in rf listen mode inadvertently. It then remembered some stray 433mhz signal which gets sent periodically turning device on and off

aksell1981, 2018-03-04:
Yes, I've figured out bymyself. I noticed switch is randomly toggled even when it's not connected to WiFi.
As you can see from my issue I tried to flush tasmota over the air first and used long press of 1st button to put it in "programm mode". With original sonoff firmare long press of 1st button as far as I understand turns the switch in "RF codes learning mode". I suspect that I recorded some noize RF signals from aerial during doing it.
This is only my guess.  
That chip you are asking is RF receiver IC. 
Data sheet is here: www.jmrth.com/download.asp?n=470R.pdf
As you can see I cut off data output from this chip. 
Sorry for my English...

FatBeard, 2018-03-04:
Ya. I think that’s what the problem is. I have an owl electric monitor which runs on same frequency. I’ve used the clear function documented on the sonoff site to attempt to fix the problem before I cut the tracks. It’s worked for the past few hours so fingers crossed. Will keep you posted. 

3.2 Clear 433MHz codes

Press and hold the touch button of T1 until you hear two beep sounds. Press any button on the RF Remote. You’ll hear a beep sound when the code is cleared successfully.

aksell1981, 2018-03-04:
Good to hear it. Looks like you solved it much easer way.
By the way I wrongly inform you. Link with datasheet pdf above is for chip which is from left side - RF receiver IC (8 legs).
From right side (16 legs, U2 on PCB) is Capacitive Touchpad IC HD6001. Datasheet is here: http://www.hab.com.tw/upload/2013879261133084.pdf

FatBeard, 2018-03-05:
Ya, a day later and it seems to have solved my problem. Thanks for your help. It drove me nuts for a few days. Thought I had a faulty device.