arendst/Tasmota #1049: I2C support multiple of the same sensor type

pascalsaul, 2017-10-22:
You can add multiple different sensor on the I2C bus but not multiple of the same sensor. On most sensor you can change the address to support multiple of the same type.

I see that the I2Cscan is detecting both sensors but not displaying the data:
14:11:50 MQT: stat/th_mv/RESULT = {"I2CScan":"Device(s) found at 0x23 0x5c"}

Is it possible to add support for mutliple sensors of the same type?

This is tested with the BH1750 but the same is valid for the BME280 and many more I guess ;)

arendst, 2017-10-24:
Adds too much extra code so won't be done.

pascalsaul, 2017-10-24:
Ai, really, that's a pitty... 

I understand you want to keep the code as clean and tiny as possible but what would add to much code?

Since the maximum addresses for a BME280 and BH1750 on the I2C bus is just two. You're scanning both addresses already in the code so is the extra code due using an array?

I can also work with statics like BH1750 #1 (0x23) and BH1750 #2 (0x5C) and BME280 #1 (0x76) and BME280 #2 (0x77) ;)

I just compare this feature with your multiple version of the DS18X20.

I don't want to push but just want to understand why this request has such a code impact :)

stefanbode, 2017-10-28:
Can you give a hint for the use-case you want to address with multiple BH1750? I created a lot of devices already and for me it always worked to have one connected to the ESP. Instead of measuring at two points and connect with long wires I just add another ESP.

pascalsaul, 2017-10-28:
Measure light on the inside and outside in a room with a BH1750. The BME280 I use for a roomtemperature reading but also measure humidity in the ventilation channel with a other BME280. Using special a ESP for just 1 sensor is overkill but putting them on a bus is far more interesting. Also I don't have every 230V or space to put a Sonoff. But I do have a telephone or data cable there. So connecting a sensor on the bus is far less complex since it's I2C.

closed 2018-05-07 by stale[bot]

demian85, 2018-12-08:
Hello, I was just trying to achieve the same but I don't know how to proceed.
I have a BME280 and a BH1750 sensor, both I2C, but only one of them is being recognized even though they are supposed to have different addresses, right?
I tried pulling up/down the address pin for both sensors and no luck.
I'd appreciate your help, thanks!

Littlesloths, 2021-04-03:
MY usecase is: to connect two SI7021 or AM2301 to read outside and inside abs humidity to control ventilation.

Littlesloths, 2021-04-16:
Culd anyone please comment if with current tasmota multiple devices on I2C one GPIO work? And still only one of the same type?

corbolais, 2023-05-17:
Use case: I need 4 HTU21 (or SI7021 for that matter) on one single device. Can be either or ESP[32]. Requirement to monitor central heating temperatures with identical sensors.

sfromis, 2023-05-18:
The use case of multiple HTU21/SI7021 I2C sensors is not relevant to this thread, as these sensors have a fixed I2C address of 0x40. In some cases, Tasmota does have support for more than one sensor of the same type, but that's only when the sensor already has a configurable address.