arendst/Tasmota #1050: counter module bug / feature request

MatzeJoerling, 2017-10-22:
Hi,

i have some issues with counter module:

issue:
when i count something (W/h impulse)  in pulse timer mode and it gets a impulse every n seconds and then it suddenly stops (powered off the tool) sonoff still reports the n seconds pulse time, forever (until i power on the tool again). 

it looks like / and counts still power for a tool that is not running.
there should be some magic thats stretches the pulse time to n seconds + Telemetry period, n seconds + 2 x Telemetry and so on (if no new pulse arises). It should go to ∞ after m telemetry periods.

feature request:
it would be nice to have a counter that sends count and  the pulse time in one message. pulse time should some average of the last n values. 
Another way would be to send telemetry (only for counter) after s counts. then i can do the math with nodered. but 5min (or less) is to long for some responsitive power meter.

regards Matze

davidelang, 2017-10-22:
see theLED* commands, if they don't work for your device, then the LED you are 
concerned about is hard-wired and cannot be controlled by software.

MatzeJoerling, 2017-10-31:
Sorry davidelang,
that doesn´t fit, this is not about LED, it is about counters.

regards

closed 2018-05-07 by stale[bot]