arendst/Tasmota #1051: bme280 reading high

Gompka, 2017-10-22:
I recently hooked up a bme280 to my wion wifi plug, but the temperature it is returning is about 10 degrees F too high. The humidity and pressure numbers seem to be correct. Is there a way to adjust the calibration to fix this?

smadds, 2017-10-22:
I have the same problem. I have mounted the BME280s so that they have no unwanted heat rising onto them, yet a side-by-side test with a DS18B20 consistently shows the BME +4degC. This is consistent over about 6 sensors.
I have just added a correction factor into the node-red flow until an explanation can be found.

```[{"id":"7f695082.866f1","type":"function","z":"d9298858.3c8cc8","name":"-4","func":"msg.payload = msg.payload - 4;\nreturn msg;","outputs":1,"noerr":0,"x":850,"y":100,"wires":[["5c3b827b.bbe3cc","8b52b366.48456"]]}]```

pascalsaul, 2017-10-22:
I use a BME280 for my roomthermostat and the temperature is correct in Celcius for me. The same is valid when I use the DS18B20. Both stand-alone mode. Hooking up a DHT22 does give me a difference of ~+4C. So I've thrown away all the DHT's since the data is not useful enough for me.

I did not compare a BME280 together with a DS18B20 on the same device within the same box. I can do that test also and see if I spot huge differences. I think you need to use the right sensor for the right application though ;)

Gompka, 2017-10-22:
I took smadds advice and adjusted the temperature in software with homeassistant. This is good enough for me.

smadds, 2017-10-23:
I've been comparing code between Tasmota's xsns_bmp.ino file and Bosch's example code and there seem to be a number of differences. I can appreciate the bitwise shifting may be more efficient than division, but... 
* the manufacturer's code includes squaring var2 variable in line 923. Is the squaring in Tasmota the same?
* also the final scaling looks different - is 
```(var1+var2)/5120 ```
really the same as 
```((var1+var2) * 5 + 128 >> 8) / 100```
? 
(If I test values of 3000 & 4000 I get results of 1.367 and 1.372)

[Bosch code bme280.c](https://github.com/BoschSensortec/BME280_driver/blob/master/bme280.c) 
From line 911:
```
static double compensate_temperature(const struct bme280_uncomp_data *uncomp_data,
						struct bme280_calib_data *calib_data)
{
	double var1;
	double var2;
	double temperature;
	double temperature_min = -40;
	double temperature_max = 85;

	var1 = ((double)uncomp_data->temperature) / 16384.0 - ((double)calib_data->dig_T1) / 1024.0;
	var1 = var1 * ((double)calib_data->dig_T2);
	var2 = (((double)uncomp_data->temperature) / 131072.0 - ((double)calib_data->dig_T1) / 8192.0);
	var2 = (var2 * var2) * ((double)calib_data->dig_T3);
	calib_data->t_fine = (int32_t)(var1 + var2);
	temperature = (var1 + var2) / 5120.0;

	if (temperature < temperature_min)
		temperature = temperature_min;
	else if (temperature > temperature_max)
		temperature = temperature_max;

	return temperature;
}
```

[Tasmota 5.8.0 xsns_bmp.ino](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_bmp.ino)
From line 247:
```
double Bme280ReadTemperature(void)
{
  int32_t var1;
  int32_t var2;

  int32_t adc_T = I2cRead24(bmp_address, BME280_REGISTER_TEMPDATA);
  adc_T >>= 4;

  var1 = ((((adc_T >> 3) - ((int32_t)Bme280CalibrationData.dig_T1 << 1))) * ((int32_t)Bme280CalibrationData.dig_T2)) >> 11;
  var2 = (((((adc_T >> 4) - ((int32_t)Bme280CalibrationData.dig_T1)) * ((adc_T >> 4) - ((int32_t)Bme280CalibrationData.dig_T1))) >> 12) *
          ((int32_t)Bme280CalibrationData.dig_T3)) >>
         14;
  t_fine = var1 + var2;
  double T = (t_fine * 5 + 128) >> 8;
  return T / 100.0;
}
```
Can anyone cleverer than me work out whether they are mathematically equivalent?

TylerDurden23, 2017-11-15:
I've noticed the same issue. This seems to be the same issues as encountered and fixed by ESPEasy:

https://github.com/letscontrolit/ESPEasy/issues/164

Frogmore42, 2017-12-02:
I did some studying of the datasheet and the solution from ESPEasy and I have an easier fix.
First needed to add a new define for the "config" register which is where you can set the measurement rate.
`#define BME280_REGISTER_CONFIG        0xF5
`
Next, I modified the setting of the config register to set the sensor to wait a second between measurements.  The data sheet says it might ignore this request when in normal mode, so I put it to sleep first.  I turned off oversampling for T, H and P, since they should not really be needed for normal environment monitoring.  Normal mode is fine for this according to the data sheet.

> 3.5.2 Humidity sensing Description: A low data rate is needed. Power consumption is minimal. Forced mode is used to minimize power consumption and to synchronize readout, but using normal mode would also be possible.  
Table 8: Settings and performance for humidity sensing
Suggested settings for weather monitoring 
Sensor mode forced mode, 1 sample / second 
Oversampling settings pressure ×0, temperature ×1, humidity ×1 
IIR filter settings filter off

I set pressure to x1 to continue to get pressure readings 

`
  if (BME280_CHIPID == bmp_type) {

    Bme280CalibrationData.dig_H1 = I2cRead8(bmp_address, BME280_REGISTER_DIG_H1);
    Bme280CalibrationData.dig_H2 = I2cReadS16_LE(bmp_address, BME280_REGISTER_DIG_H2);
    Bme280CalibrationData.dig_H3 = I2cRead8(bmp_address, BME280_REGISTER_DIG_H3);
    Bme280CalibrationData.dig_H4 = (I2cRead8(bmp_address, BME280_REGISTER_DIG_H4) << 4) | (I2cRead8(bmp_address, BME280_REGISTER_DIG_H4 + 1) & 0xF);
    Bme280CalibrationData.dig_H5 = (I2cRead8(bmp_address, BME280_REGISTER_DIG_H5 + 1) << 4) | (I2cRead8(bmp_address, BME280_REGISTER_DIG_H5) >> 4);
    Bme280CalibrationData.dig_H6 = (int8_t)I2cRead8(bmp_address, BME280_REGISTER_DIG_H6);
    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0x00);      // sleep mode since writes to config can be ignored in normal(Datasheet 5.4.5/6 page 27)
    // Set before CONTROL_meas (DS 5.4.3)
    I2cWrite8(bmp_address, BME280_REGISTER_CONTROLHUMID, 0x01); // 1x oversampling
    I2cWrite8(bmp_address, BME280_REGISTER_CONFIG, 0xA0); // 1sec standby between measurements (to limit self heating), IIR filter off
    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0x27);      // 1x oversampling, normal mode
 } else {
    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0xB7);      // 16x oversampling, normal mode (Adafruit)
  }
`

This window does not appear to like my code, but the important thing is to set register F5 "config" to  0xA0 to limit the self heating.  Reducing the oversampling is probably a good idea too (the two control registers).

Frogmore42, 2017-12-02:
Before I made this change, my two sensors that are on the same breadboard right next to each other and measuring the same air always showed the BMP about 1.5 deg F higher.  After making the change, the BMP now reads about 0.5 deg F higher, which is well within the spec limits.

Frogmore42, 2017-12-02:
@smadds you might want to look at section 4.2.3 of the Bosch BME280 manual, that has the integer version of the conversion formulae.  You are looking at the floating point implementation which is only recommended for use on a PC.

WouterNefkens, 2018-04-10:
`if (BME280_CHIPID == bmp_type) {  // #1051
 +    Bme280CalibrationData.dig_H1 = I2cRead8(bmp_address, BME280_REGISTER_DIG_H1);
 +    Bme280CalibrationData.dig_H2 = I2cReadS16_LE(bmp_address, BME280_REGISTER_DIG_H2);
 +    Bme280CalibrationData.dig_H3 = I2cRead8(bmp_address, BME280_REGISTER_DIG_H3);
 +    Bme280CalibrationData.dig_H4 = (I2cRead8(bmp_address, BME280_REGISTER_DIG_H4) << 4) | (I2cRead8(bmp_address, BME280_REGISTER_DIG_H4 + 1) & 0xF);
 +    Bme280CalibrationData.dig_H5 = (I2cRead8(bmp_address, BME280_REGISTER_DIG_H5 + 1) << 4) | (I2cRead8(bmp_address, BME280_REGISTER_DIG_H5) >> 4);
 +    Bme280CalibrationData.dig_H6 = (int8_t)I2cRead8(bmp_address, BME280_REGISTER_DIG_H6);
 +
 +    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0x00);      // sleep mode since writes to config can be ignored in normal mode (Datasheet 5.4.5/6 page 27)
 +    // Set before CONTROL_meas (DS 5.4.3)
 +    I2cWrite8(bmp_address, BME280_REGISTER_CONTROLHUMID, 0x01); // 1x oversampling
 +    I2cWrite8(bmp_address, BME280_REGISTER_CONFIG, 0xA0);       // 1sec standby between measurements (to limit self heating), IIR filter off
 +    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0x27);      // 1x oversampling, normal mode
 +  } else {
 +    I2cWrite8(bmp_address, BME280_REGISTER_CONTROL, 0xB7);      // 16x oversampling, normal mode (Adafruit)
 +  }`

Why changing the register config only for when BME280_CHIPID == bmp_type?

majherek, 2018-04-10:
Hi,

I have still problem with my BME280.

```
tele/sonoff9/SENSOR {"Time":"2018-04-10T08:38:07","DS18B20":{"Temperature":20.3},"HTU21":{"Temperature":21.8,"Humidity":40.0},"TempUnit":"C"}
tele/sonoff8/SENSOR {"Time":"2018-04-10T08:38:46","Switch1":"ON","BME280":{"Temperature":22.2,"Humidity":42.2,"Pressure":1000.2},"TempUnit":"C"}
````
I have 2 deg C different between DS and BME. 
I compare 5 types of sensors (HTU, BME, Xiaomi, Oregon and DS). Xiaomi and Oregon are 0,4 deg C higher than DS, HTU21 is 1,5 deg C higher than DS and BME280 is 2 deg C higher than DS.  HTU, BME and DS are connected to two Wemos D1 mini with tasmota 5.12.0i.

Does anyone still have a similar problem?

WouterNefkens, 2018-04-10:
Yes still same problem, also around 2 degrees difference.

Frogmore42, 2018-04-10:
I only had a bme280, but Theo generalized the code when he added it.

Unless the sensors are in a temperature controlled medium that is very uniform, they will read different values.  It is easy for a difference of a few inches to make the difference you are seeing.

Remember the old saying, a person with one clock always knows what time it is.  A person with two is never really sure.

bontas69, 2019-07-20:
What can be done to correct the temperature redings from bme/bmp sensor.I have 6 degree offset. Seems that arendst give the solution here :https://github.com/curzon01/Sonoff-Tasmota/commit/ca1e5f85ccbfdddfa6c0e95404df7ed2614e8b57 but how can the code be uploaded to sonoff? Some copy-paste in the console in web page? Please some advice or tuturial...
[image]

meingraham, 2019-07-21:
This fix is in the Tasmota firmware since long ago. Your sensor may be inaccurate. You will have to account for the discrepancy and adjust your use of the reading accordingly.

bontas69, 2019-07-21:
Thanks for the answer.I believe forced mode is mandatory for using bme/bmp sensor as weather station.,reading temperature once/minute.... from some bosch recomendations:
 (According to Bosch Sensortec the suggested settings for climate /
weather monitoring are:

Sensor mode:  forced mode, 1 sample / minute
Oversampling settings: pressure ×1, temperature ×1, humidity ×1
IIR filter settings: filter off

To measure new values in the forced mode a write to the
"BME280_REGISTER_CONTROL" is necessary, which I added to the temperature
reading.)
On tasmota sampling is once/second ,heating sensor with hot gun, the temperature changes in webinteface every second or two.
Is there any serial or console command to modify sampling rate of bme/bmp sensor, experimental from 1sample/second to 1sample/minute, to choose a useful and error free position?

meingraham, 2019-07-21:
Sampling rate for each kind of sensor is hard coded in the sensor's driver and is set according to "optimal" sampling rate for the sensor based on testing. YOu would have to edit the driver and compile the firmware with your modifications.

Frogmore42, 2019-07-21:
A year ago, I ran into this problem.  I read the data sheet and discovered that the sensor was configured in a way that ensured self-healing.  I reconfigured the settings to eliminate it, by reducing the sample rate to 1/second.  This eliminated the self heating for many.  Theo took the change I made and generalized it so everyone could benefit from it.

The Bosch recommendation for weather monitoring is not to help with self heating, but to help with power consumption (of course, the two are related).  While it is possible that you need to reduce the sample rate even more, it is more likely there is something else wrong.  As the old saying goes, "a man with one clock always knows what time it is, a man with two is never really sure."

It is possible one or both of your sensors are calibrated incorrectly and/or broken.  I know many people have gotten defective DS18B20 sensors.  I also know that both of these sensors are very sensitive and that there can be temperature gradients.  To eliminate the gradients, put the sensors almost touching and then use a small fan to blow air on them (small computer fans are great for this).  This should eliminate and temperature gradients.  It would also reduce self heating, so also try it without the fan.  If you have a thermal imager use that 6 degrees is really easy to see with one of them.  That would tell you is the temperature really is higher/lower or if the sensor(s) are more likely just out of calibration.

joba-1, 2019-07-21:
I have many BME280/BMP280 and did some testing with 1/s and 1/min. 
Didn't get a difference (or maybe 0,1°C). 

All of them tend to show the same temperature, but 1-2° higher compared to other sensors I have though.

bontas69, 2019-07-21:
Thanks for the answers.The correct temperature is that of DS18B20 sensor. Maybe the BMP280 is defective,I don't have one more bmp to test it.Anyway, 6 degree misreading is outrageous.Would be useful a script /command to adjust sampling rate of the sensor or some parameter to adjust via console

Frogmore42, 2019-07-21:
You aren't going to see a 6 degree rise from self heating.  Clearly your sensor is defective.  While it might be useful for some scenarios to have more control, that means more things for the average user to get wrong.

If you feel strongly enough about the need for control, you can fork the code and make the change.  I don't think there is anyone else that feels the need for this, so it probably won't happen soon, or ever, otherwise.

meingraham, 2019-07-21:
@bontas69 I recently had this dialog with Theo. As each sensor requires tuning and calibration, providing a command to handle this "generically" will likely lead to more issues for users who have not researched it properly. Fixes are incorporated into the driver as issues are uncovered. But changing the settings via a runtime setting is, at this time, not going to be added. However, that does not disallow anyone from customizing the driver code for the sensor in question and compiling one's on version to suit their requirements... having done the necessary research to understand the sensor properly.

bontas69, 2019-07-21:
Thank you for your support.Definitely I will not write a code for calibrating bme sensors because is way above my area of expertise. I"m just an engineer,electronics,with some interest in automation,all I can do is to copy-paste a command in terminal,I will not write any code or made custom firmware.The bme wrong temperature issue is all over internet, especially on alternative firmware esp8266.Here is described the overheating, oversampling, high update_interval and the wrong power mode: https://github.com/esphome/issues/issues/402. 
[image]
[image]
The bme/bmp are cheap and powerful, mine may be defective,some one said there is no difference between 1sample/sec and 1sample/min. Maybe on lower temperatures, on winter,I don.t know.
If I could , I will have the feature of a simple thermostat in webgui,or al least a firmware with thermostat embedded,especially for sonoffs with temperature sensor attached.
Many users demands for thermostat sonoff, not implemented by complicated rules, hard to maintain or modify.

kolonka, 2019-09-05:
Hey,
I got 4 of the bme280 sensors with an integrated vr over here. I programmed them on Atmega chips and they were running fine. I hooked up one to tasmota and noticed, that the values are like 4° higher then on my self-made boards. I bought 3 additional sensors without integrated vr, since they are potentially more accurate because the vr can put heat on the pcb.  The tasmotas are reading 26.2-28.3° with the sensors without vr right now. One tasmota is located 1m away and has a sensor with vr. It is reading 27.1°. My self-made board is located next to the 3 tamotas and reads 24.81°.
Since I have at least 'some' experience with the sensors, I could have a look into tasmota code. Only if there is a chance to get the code published into release in future. I never worked on such a project. Do I have to clone the repository, work on it on a new branch and then submit a merge request with my commits?
For general question: Why don't you just fetch new sensor data on Mqtt-Update? That would make the update rate dependent on TelePeriod. Since TelePeriod is at least 10s, the sensor would not have any self-heating issues. Btw: In my code I am running the chip in forced mode, about 2s interval and 4x filter.

matteius, 2022-10-24:
I ran into the issue where the BME280 sensors read about 5.5 deg C higher than the ambient temperature and found this issue and several other issue reports across projects.  I first worked around it in code with adding a temperature offset, but then I noticed that in sleep mode it reads the correct temperature and in forced mode it reads the correct humidity and pressure.   I am using the basic BME280 python library and then also have working the pimironi version of BME280 -- its in this library that I modified the usage of the chip to do the following:

1. Call setup method and ensure that setup method and ensure that method resets the chip, example: `self._bme280.set('RESET', reset=0xB6)`
2. Initialize the chip to sleep mode for reading the temperature
3. Read the Temperature
4. Reset the chip again  `self._bme280.set('RESET', reset=0xB6)`
6. Initialize the chip to forced to read the humidity/pressure.
7. Read the Humidity and Pressure
8. Repeat

So far this has been the most accurate readings I've been able to get that match my ThermPro gauges.  It clearly isn't heating up 5+ degrees Celsius within the first seconds of being on because this logic keeps the sensor active and reading it every 10 seconds and getting the same results, but it does seem that having the sensor in forced mode causes it to read a much higher temperature, which is odd since forced mode will "perform one measurement, store results and return to sleep mode"

```
    def update_sensor(self):
        self.setup(mode="sleep", temperature_oversampling=1, pressure_oversampling=1, humidity_oversampling=1, temperature_standby=1000)
        
        self._bme280.set('CTRL_MEAS', mode="sleep")
        while self._bme280.get('STATUS').measuring:
            time.sleep(0.001)
        
        raw = self._bme280.get('DATA')
        self.temperature = self.calibration.compensate_temperature(raw.temperature)

        self.setup(mode="forced", temperature_oversampling=1, pressure_oversampling=1, humidity_oversampling=1, temperature_standby=1000)

        self._bme280.set('CTRL_MEAS', mode="forced")
        while self._bme280.get('STATUS').measuring:
            time.sleep(0.001)

        raw = self._bme280.get('DATA')

        self.pressure = self.calibration.compensate_pressure(raw.pressure) / 100.0
        self.humidity = self.calibration.compensate_humidity(raw.humidity)
```

AnDrEaZaNoLeTtI, 2024-10-18:
I know that is an old topic... but I hope I can be useful to anyone who has the same problem of sensor overheating...

I recently created a humidity detection node using an ESP8266 and a BME280 and I ran into a problem of high temperature read by the sensor. my readings were 6-8 degrees higher than actual.

I read many forums that talked about self-heating problems and I resigned myself to using another sensor for temperature readings and using the BME280 only for humidity.

almost by chance I noticed that sometimes for a few seconds the readings of its sensors aligned and upon investigation I discovered that this happened when the wifi signal was lost.

i modify firmware in order to turn on/off wifi according with a input pin and i connect my laptop with esp8266 throw serial pin in order to read logs. I discovered that with wifi off the readings are close up to 0.1°C, with wifi on the error was 7°C.

the problem was the sensor being too close to the esp8266 antenna... I lengthened the wires to keep the sensor at least 15 cm away from the antenna and I no longer had problems with high temperature readings even with reading time of 10s... with much faster readings (1 s)  there is still an overtemperature, but limited to around 1.5 °C

Gompka, 2024-10-22:
> I know that is an old topic... but I hope I can be useful to anyone who has the same problem of sensor overheating...
>
> I recently created a humidity detection node using an ESP8266 and a BME280 and [...]

Interesting, is there a way to program esp8266 to turn off the wifi for measurements? I have my bme280 inside a wifiplug so no way to keep it 15cm away from the wifi antenna.