arendst/Tasmota #1052: All status

smadds, 2017-10-22:
Enhancement request:

Would it be possible to add a status call that outputs all of the status at once as a single json hierarchy?

I realise the status 0 command outputs this sequentially, it's just messy then to build a comprehensive status page. Not too bad if you have just one or two devices, but a pain if you have a lot more. Is it a resource issue that prevents it being a single object?

davidelang, 2017-10-22:
On Sun, 22 Oct 2017, Simon wrote:

> Is it a resource issue that prevents it being a single object?

yes, mqttmaxmessagesize is already a problem, and ram is very limited, so e cn't 
put everyting into one message.

Hollako, 2017-10-22:
Can you please advise how to parse a value using an JSONPath from the below HTTP reply?

STATUS11 = {"StatusSTS":{"Time":"2017-10-22T20:40:45", "Uptime":0, "Vcc":3.192, "POWER1":"ON", "POWER2":"OFF", "Wifi":{"AP":1, "SSId":"Wifi LAN", "RSSI":44, "APMac":"98:DE:D0:D8:6C:FD"}}}

I am trying on the JSONPath online extractor it keep giving me error but when i deleted the "STATUS11 =" it works. However the HTTP reply always has the "STATUS11 =" work prior to JSON data.

Thank you,

smadds, 2017-10-23:
Not sure what that has to do with this ticket, @Hollako, but the json part of the response surely starts with the opening "{" and ends with the closing "}". How you extract that from the http response would depend on your programming language. Google is your friend if you are new to the language!

arendst, 2017-10-25:
@smadds Simon, as David already noticed concatenating all status messages in one message means that the mqtt buffer needs to be extended taking up too much "precious" RAM memory only to be used for this long message as it can not be released by the PubSubClinet library.

That's also the reason why command ``gpios`` responds with several mqtt messages...

closed 2018-05-07 by stale[bot]