arendst/Tasmota #1055: POWER ON vs POWER1 ON

flexiti, 2017-10-22:
If we have one relay than we have : POWER ON/OFF, but if there more than one we have POWER1 ON/OFF.
It would not be better to always be POWER1 ?

davidelang, 2017-10-22:
not for backwards compatibility

Changing this would break people's configs.

now, a config option to do this would be good, but not as the default.

jneilliii, 2018-01-12:
I agree with @flexiti, it seems like having a consistent response no matter what module type you have would make a whole lot more sense.  I've recently come to deal with this issue with my [OctoPrint-Tasmota](https://github.com/jneilliii/OctoPrint-Tasmota) and [OctoPrint-TasmotaMQTT](https://github.com/jneilliii/OctoPrint-TasmotaMQTT) plugins that for certain devices sending command Power1 returns the result {'POWER':'ON"} vs {'POWER1':'ON'}. Of course I can get around it in my code by leaving the relay index blank, but it has caused some headaches along with the web response differences of tasmota versions older than 5.11. 

Would it not be possible that if the request is for POWER1 that it returns POWER1 and if the request is for POWER that it returns POWER?  That seems like it would handle any backwards compatibility issues.

jneilliii, 2018-05-13:
Keep alive, never got a response.

ascillato, 2018-05-13:
If you type on the console `POWER ON` or `POWER1 ON`, both works. So this issue has been solved some time ago.

Please close this issue.

jneilliii, 2018-05-13:
It's a matter of integration and consistency IMO.  If you send a command to `POWER1` you would expect to receive a stat response for `POWER1`, but in the current state (as of 5.12.0) the response is saying `POWER` is on. Yes, the command still works, but the response is unexpected.  This is only true in the single relay device models, so for example on a sonoff dual sending a `POWER1` command returns a stat response for `POWER1` and a `POWER2` command returns a stat response for `POWER2`.

ascillato, 2018-05-13:
I understand, can you @jneilliii do a PR for that?

jneilliii, 2018-05-13:
I've never coded arduino before but will look into it.

arendst, 2018-05-13:
@jneilliii no need.

I'll look into it. The "problem" is as stated in the first response related to legacy usage. I'll try to change it as requested but it's not that simple as an index of 1 is always present even when a user does not enter it. Only the amount of relays available  currently decides if a postfix is used or not.

I'll need to redesign things for this...

arendst, 2018-05-13:
As I said before there is more to it that meets the eye too....

When using command ``power1`` I would expect to also see "POWER1" in the status messages and blink messages and all other messages where the power status pops up. So i prepared a persistent variable to store this option.

At first I thought it would be nice to have this option hidden and only activated if a user executed a command like ``powerx on/off/toggle`` but then if a user by accident executes the command ``power on/off/toggle`` all future status messages would return just power without the index....

So my solution is providing another SetOption allowing a user to select legacy (without the 1 index) or fixed to use indexes even if the user by accident enters ``power`` when ``power1`` option is enabled.

Some people might disagree but as others planning to ues rules it makes sense to have a known event like "POWER1#state" in all cases.

About to release now.

localhost61, 2018-05-13:
Is it SetOption16 or SetOption26 ???

arendst, 2018-05-13:
See commands in wiki. It's 26.

DavidFW1960, 2018-05-20:
I just updated to 5.14 and I have never SetOption26. I edited my configuration in Home Assistant to use power1 instead of power and my switches stopped working. I edited configuration back to use power as I had it when I first flashed Tasmota and they are working again. So I'm confused by this. I am using 2 of the S22 switches.

jneilliii, 2018-05-20:
My understanding is that if you want to use the new 'power1' syntax you have to SetOption26, otherwise the previous syntax remains as before and returns 'power'.

DavidFW1960, 2018-05-20:
Yes I see that now. Thanks.

ascillato, 2018-06-02:
Issue Solved. Closing. If you find any other issue, please open a new one. Thanks.

closed 2018-06-02 by ascillato