arendst/Tasmota #1058: SonOff TH WDT Reset

daspid92, 2017-10-23:
Hi, I've a problem with a SONOFF TH, I'm able to flash it, but when I try to change model or do everything else which requires a restart (but sometimes also randomly) in console i see this:
`20:48:33 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
`
And the module won't start anymore, the only way to make it start again is to flash it again, what can be? (I've only added the 2 resistors to use GPIO4)

ionciubotaru, 2017-10-25:
I had the same problem with a SONOFF BN-SZ01 - I cannot fix it and conclude its a hardware problem

jetema, 2017-10-25:
In the case you reflash again the sonoff don't forget ERASING COMPLETELY the flash memory BEFORE, it only takes 3 seconds.
Try as well update or downgrade your version of tasmota the aim of this, is run Tasmota in a different part of the flash memory
for more information about map of memory 
see this -->  https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

pascalsaul, 2017-10-29:
How can I erase the flash memory complete??? I've bricked three basic's now I guess. It was flashed with a version from a friend and I reflashed it with my 5.8.0o with all sensors enabled (~476kb) and flash was after several tries successfully but after that the basic is continues resetting and can't get it into bootmode anymore. I think there is a serious problem when you reflash Tasmota with a different version number through serial? I thought you could go always in to bootmode to reflash it but now I got 3 basic's that won't anymore? @arendst do you know what happened perhaps? Tried to reflash it with a smaller version without additional sensors but I'm not able to anymore...

davidelang, 2017-10-29:
no matter what you flash, you cannot flash something that will prevent you from 
getting to the bootmode.

to erase the flash, create a file with all zeros, and flash that.

But you should not need to do that, at most you just change the CFG_HOLDER 
variable so that the firmware doesn't consider whatever is in the config section 
of the flash to be valid

pascalsaul, 2017-10-31:
I've updated the Wiki a bit: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting

I can't get the devices erased good or flash again:
%USERPROFILE%\.platformio\python27\Scripts>esptool --port COM12 write_flash --flash_mode dout --flash_size 1MB 0x0 firmware.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 475008 bytes to 327693...
Writing at 0x00008000... (14 %)
A fatal error occurred: Timed out waiting for packet header

daspid92, 2017-11-10:
In my case I've found the problem, I've removed the 10k pull-up and 0k resistors for GPIO4 and it started working correctly, than I've added again the 10k resistor and soldered a wire instead the 0k resistor not connecting it to the sensor jack, in this way it works...

pascalsaul, 2017-11-17:
Yes, but I've this with the basics... I'm waiting for a new FTDI adapter...

Mak69, 2018-04-06:
Maybe a little bit late, but just uploaded tasmota firmware on a new Sonoff 4ch.
Same issue, always "wdt reset" and no way to connect to the web page.
For me it was just a current problem, maybe not enough from my usb FTD232 adapter.
Compiled with a fixed IP (optional), disconnected from USB adapter (for safety reason) and plugged on main 220Vac.
It started straight forward and I've been able to connect to its web page.

dweston, 2018-04-19:
I've been trying to flash a Sonoff TH and encountered this problem. I've tried both the master branch v5.12.0 and dev branch v5.12.0m. In both cases I'm encountering the:
` ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset`
issue with every restart. The code flashes OK. All functionality appears normal upon first start but after any reset - cactus.

dweston, 2018-04-19:
Yikes - exactly the same symptoms with a Sonoff POW. I'll try and earlier version.

dweston, 2018-04-19:
Same problem with v5.11.1

Must be a hardware issue I've created. Both devices were working previously off code flashed using platformio home v 0.9.5 core 3.5.3a9

Since then I've been trying to get OTA to work, originally with partial success using the web update & file update but requiring odd sequences of flashing minimal firmware followed by OTA web generic, followed by firmware complied locally. No sure what's changed now.