arendst/Tasmota #1059: Sonoff Basic no Reaction

jannnfe, 2017-10-23:
Hello,
I have a Sonoff Basic, which reacts to nothing more. When I connect it via the 3.3V or the normal 230V port, it reacts to nothing. After the soldering I have it connected via GND and 3.3V connection, the green LED has also blinked for 30 sec, then I have the power connection interrupted to come into the flash mode and then the Basic responds to nothing more. Did I possibly soldered incorrectly and it came to a short circuit? Can soldering current flow? The serial adapter was set to 3.3v and the pins were correctly assigned. I fear I have the Sonoff broken, because it is not synonymous with the normal 230v port reacts and still the normal firmware is installed.
Thanks!

jetema, 2017-10-24:
Maybe your sonoff is broken/bricked, anyway if you want flashing without soldering have a look at this issue 
https://github.com/arendst/Sonoff-Tasmota/issues/792

jannnfe, 2017-10-29:
I think I've found the problem. The problem was not soldering but computer. On my MacBook, the flashing did not work properly. I just flashed again the same board with a Windows PC and now it works again! Thanks anyway for your help!

closed 2017-10-31 by jannnfe