arendst/Tasmota #1060: SOnOff 4CH - Configure one external switch to send to a different topic - help wanted

sheppy99, 2017-10-24:
I have a SOnOff 4Ch flashed with Tasmota 5.0.0n and want to use 3 external switches. I would like 2 of them to simply toggle the relays, and 1 of them to send a MQTT message to another item. Without doing anything the switches toggle the relays, so I just need to make one behave differently. Is there a way to do this? If not can it be added soon?

davidelang, 2017-10-24:
it can't be done today, and probably won't be done soon.

you can make all the switches send messages, but not just some of them. This 
would be a pretty drastic change and eat up valuable code space for a corner 
case.

sheppy99, 2017-10-24:
Is there a way of remapping what the individual item that the switches send to MQTT is. For example:
Switch 1 = cmnd/mySOnOff/Power3 - switches relay 3 via MQTT subscribe
Switch 2 = cmnd/mySOnOff/Power4 - switches relay 4 via MQTT subscribe
Switch 3 = cmnd/mySOnOff/Power5 - doesn't switch anything as relay 5 doesn't exist, but can be mapped to an item in OpenHAB
With this arrangement the standard publish mechanism could be used?

calcipher49, 2017-10-24:
Why switches for toggle relays? It's better to use buttons and reserve switches for MQTT messages.

sheppy99, 2017-10-24:
Because unused light switches already exist on the wall and I want to use them to send MQTT messages

calcipher49, 2017-10-24:
'Because unused light switches already exist on the wall ' 
Using soldering, you can wire existing swtches in parallel to buttons.

sheppy99, 2017-10-24:
I want to use 3 switches, 2 of them to operate switches on my SOnOff 4Ch, and one to send a MQTT message to OpenHAB to operate a different relay on another device. I already have the 2 working with the existing switches, it’s the third one that I need to work out

calcipher49, 2017-10-24:
Maybe you can configure the first 2 as buttons (and internal toggle relay) and third as switch, with SwitchTopic enabled...

sheppy99, 2017-10-24:
I’ll look into that tomorrow - thanks.
It’s late in this part of the world but please keep the suggestions coming as I’ll look at them in the morning

davidelang, 2017-10-24:
On Tue, 24 Oct 2017, Kevin wrote:

> Is there a way of remapping what the individual item that the switches send to MQTT is. For example:
> Switch 1 = cmnd/mySOnOff/Power3 - switches relay 3 via MQTT subscribe
> Switch 2 = [...]

your home automation can receive the messages and issue whatever commands you 
want to take place.

This wouldn't be power3/4/5, it would be power1/2/3 (the topic would be 
different than the one used by the buttons/relays)

sheppy99, 2017-10-24:
Ideally I'd like to operate some of the lights even when the Home Automation is down which is why I'd like the first 2 switches to operate relays directly. I'll experiment to see what I can work out.

sheppy99, 2017-10-25:
It seems that the only way I can get an MQTT signal out of one of these light switches is to leave a relay channel unused and then use that as a trigger for OpenHAB. Selecting "relay5" as its target doesn't provide anything on MQTT if I turn on SwitchTopic. I guess for now that's as good as it gets if I want to retain functionality for if OpenHAB is down.

One other thing I've discovered for anyone following on is that this module (SOnOff 4Ch) hates switching wall warts On and Off and repeatedly crashes if you do, likely due to a combination of inrush current spike and wiring resistance. Fortunately I've found a switched mode PSU from an old router that has a soft start function that doesn't upset it, and I've ordered a soft start module from AliExpress for my other light transformer

closed 2018-05-07 by stale[bot]