arendst/Tasmota #1061: mqtt qos setting

apathyuk, 2017-10-24:
Hola

How do I set the QoS for MQTT?  Looking at the docs, it says it is done as part of the topic subscription but I cannot see an option in Tasmota.

I am using mosquitto and v5.7.0 of tasmota.  

Any pointers?

James

davidelang, 2017-10-24:
what are you trying to achieve with your QoS settings?

ionciubotaru, 2017-10-25:
You can edit the mqttpublish function or you can set retain flag
qos 2 is not supported by the mqtt library

stefanbode, 2017-10-28:
Agree, especially locally the should be no situation where a repeat of the message is required. If you want to overcome a network tempering attack this also will not help you because there is no guarantee that the Jammer will ON all the time. These kind of attacks must be solved in other ways.

apathyuk, 2017-10-28:
I have one Tasmota that is further away from the house and therefore the WiFi signal is less strong. I am looking at improving this but hoping that QOS would make it more reliable.

stefanbode, 2017-10-28:
Ok, make sense. Had a similar problem some time ago. Anyhow I must admit, that the pubsubclient library or maybe the ESP itself has problems with the subscription when messages are common very fast one after another. Therefore my change to QOS 1 on publish and subscribe at the end did not solve the problem. It make it even worse because of the communication limits of the pubsubclient. I invested a day to replace the library with a more modern one (using much more memory), but the issues to work on messages comming in in a very short time persists. I found one article, that described that this is a limitation of the ESP and not of the pubsubclient.

To reproduce just retain some messages to the mqtt cmd topic like: Webserver 2, status 0, teleperiod 30 and so on and see what gets executed on startup. It should execute ALL, but practically it executes only one, if you have luck. At the mosquito server you see that all messages are sent. Problem is the ESP8266.

davidelang, 2017-10-28:
QOS would have no effect on a problem like this. QoS rules only come into play 
when you have direct contention, and they don't work well on wireless. They do 
nothing to address a weak station.

closed 2018-05-07 by stale[bot]

mrbgfromv, 2018-12-15:
+1
This should be user configurable to handle network outages as shown in [https://www.youtube.com/watch?v=fwb5YAPzPGk] by Andreas Spiess  at approx 08:00 min.
So if the network comes up again tasmota clients would change/"sync" their status correctly.

lucafolin, 2018-12-19:
+1
I have installed tasmota 6.3 on my client hardware but I'm  not able to set mqtt = 1.
How and where can I modify Tasmota code to gain this feature ?

tyriis, 2019-02-08:
+1
its really useless without QOS setting :(

c0r3xxx, 2019-03-16:
+1
I also would be happy about this option!

3s1d, 2020-02-03:
+1 same here.
Is there still not possibility to set Qos=1 for subscribed topics?
This would be very handy.

ascillato, 2020-02-03:
Hi, Sorry. QoS = 1 is not supported by the mqtt library (pubsubclient) used by Tasmota, so this request is not going to be added. Sorry.

imkebe, 2021-01-28:
Is there a general constraint of the Tasmota architecture that prevents adding QoS? Appart of firmware size?