arendst/Tasmota #1064: OTA upgrade fails to install "sometimes"

qorron, 2017-10-24:
I have about 30 sonoff touch and some other sonoff modules installed in my house.

to keep all in line I have written a discovery agent that flashes unconfigured modules and prompts for a name. that list can be forged in a config for an other script that upgrades every configured module and brings the config up to date.
while doing this, I noticed, that some modules refuse to take the ota update.

`cmnd/DVES_8540B7/Upgrade 1
stat/DVES_8540B7/RESULT {"Upgrade":"Version 5.7.1 from http://192.168.2.2:80/api/arduino/sonoff.ino.bin"}
`

then there is no "update successfule, restarting" line. they just reboot coming back online with the old version.

interesting:
`tasmota/balkon_decke/tele/INFO3 {"RestartReason":"Blocked Loop"}
`
as of now, all modules have managed to get to 5.8.0 except for one. that stays at 5.7.1 even after 6 or 7 updates.
others took 2 or 3 to finally get there.

that one was programmed with 5.7.1 via serial.

any ideas?

ionciubotaru, 2017-10-25:
I had the same problem in the past.
Solved using a good wifi coverage (more than 70% signal)

qorron, 2017-10-25:
that explains a lot, the module in question has:
`"RSSI":36`
it's the lowest of all.
thank you, @ionciubotaru !

closed 2017-10-25 by qorron