arendst/Tasmota #1065: CapSense Library

jtroberts1, 2017-10-24:
So I can easily use a ttp223 touch module with Tasmota to turn a light on or off.  However I was hoping to use Tasmota to turn any ordinary metal lamp into a capacitive switch without the use of the TTP223.  IE: I want to use a 1M ohm resistor to sense the change in capacitance to turn the lamp on or off.  

I guess this is more of a feature request (unless it can already be done) to add capsense library to Tasmota.  something where we can choose the send and receive pins and define a value to action a switch/relay?

Thoughts?  PS: keep up the great work @arendst

jtroberts1, 2017-10-27:
bump

@arendst   Any thoughts on implementing capsense library in Tasmota?

stoehrmark, 2017-12-08:
1+ this would be great

closed 2018-05-07 by stale[bot]