arendst/Tasmota #1066: SwitchTopic don't send MQTT

junk-barkass, 2017-10-25:
When I set SwitchTopic to for examples sonoff it does not send an mqtt message.
I'm using mosquitto_sub -v -t cmnd/# -t 8/# -t stat/# -t knapp8/# -t tele/# -t /sonoff/#  .....

See below: 
8/8/RESULT {"SwitchTopic":"sonoff"}  This was set by console 

8/8/RESULT {"POWER":"ON"}      Button pushed
8/8/POWER ON
8/8/RESULT {"POWER":"OFF"}     Button pushed again
8/8/POWER OFF

__________________________________
Tasmota version: 5.8.0n
MQTT Topic  8
MQTT group topic 8
Full topic    8/8/cmnd/

If this is not a bug, sorry and correct me. I'm trying to detect button press and then from server side control the relay. 
Best Regards Magnus

ionciubotaru, 2017-10-25:
switchtopic is used for switches not for buttons
for buttons you must use buttontopic

junk-barkass, 2017-11-03:
Yes, I did understand that later but din't get it to work. But it do work now with the latest version. Thanks for a great Tasmota!!  By the way how do I configure my sonoffs to be quiet not sending something every minute like: 
00:39:59 MQT: knapp/2/STATUS10 = {"StatusSNS":{"Time":"2017-11-04T00:39:59", "DHT22":{"Temperature":23.7, "Humidity":49.8}, "TempUnit":"C"}}
00:40:00 MQT: knapp/2/STATUS10 = {"StatusSNS":{"Time":"2017-11-04T00:40:00", "DHT22":{"Temperature":23.7, "Humidity":49.8}, "TempUnit":"C"}}
00:40:00 MQT: knapp/2/STATUS10 = {"StatusSNS":{"Time":"2017-11-04T00:40:00", "DHT22":{"Temperature":23.7, "Humidity":49.8}, "TempUnit":"C"}
or
00:30:03 MQT: knapp/1/RESULT = {"POWER":"OFF"}
00:30:03 MQT: knapp/1/POWER = OFF
00:35:03 MQT: knapp/1/RESULT = {"POWER":"OFF"}
00:35:03 MQT: knapp/1/POWER = OFF

This is for me just junk information. My server will ask for information when needed. Is it possible to disable this type of messages or increase the time between them?

Best Regards Magnus

ionciubotaru, 2017-11-04:
set a very long teleperiod

junk-barkass, 2017-11-05:
Aha. I th lugnt 0 Disabled sending. 
Thx 

Med vänlig hälsning, Magnus

ionciubotaru, 2017-11-05:
telePeriod cannot be shorter than 10 seconds

closed 2018-05-07 by stale[bot]