arendst/Tasmota #1068: GPIO14-Switch

mittelfeldmotor, 2017-10-25:
I configured a sonoff basic with gpio14 as switch01. It works, but wenn i turn the switch off for longer time, the sonoff needs few seconds to toggle. Is this normal?

Gompka, 2017-10-25:
I have experienced the same issue, it seems like the sonoff takes a moment to wake up and actually turn on my light.

jetema, 2017-10-25:
Have a look at "PulseTime" command, see if its value is zero or not

mittelfeldmotor, 2017-10-25:
where?

mittelfeldmotor, 2017-10-25:
can I find it

mittelfeldmotor, 2017-10-25:
found it. It´s zero

Gompka, 2017-10-25:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands    

You can enter the commands in the console on the webpage of your sonoff. 

Jesus, mine is set to 0, i am using the switch as a wall switch not a momentary.

jetema, 2017-10-25:
Have you Tasmota 5.8.0 or older versions ?

mittelfeldmotor, 2017-10-25:
5.4.0

jetema, 2017-10-25:
If so try a older one, you have all versions available on 
https://github.com/arendst/Sonoff-Tasmota/releases

Gompka, 2017-10-25:
I am on version 4.0.5

jetema, 2017-10-25:
Try to update to 5.8.0 via OTA, is quick and easy

Gompka, 2017-10-25:
I tried updating, but it doesn't want to take, even with use_minimal uncommitted. I will try to do it one more time and if not ota will flash manually and report the results.

jetema, 2017-10-25:
There are two methods to update, the easiest one is updating by file upload, select  sonoff.bin  ( NOT minimal )

mittelfeldmotor, 2017-10-25:
Update minimal finished, nut now sonoff is not reachable in wlan

mittelfeldmotor, 2017-10-26:
I tried https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage to config the wlan, but no chance, I think I should reflash

Gompka, 2017-10-26:
I upgraded to the latest version of tasmota and so far i have not experienced any delays with the external switch. I will check again today after work to make sure, but i think that was my issue.

mittelfeldmotor, 2017-10-27:
Upgrade to latest Version solved the problem for me

closed 2017-10-27 by mittelfeldmotor

joaohess, 2017-12-08:
I'm up-to-date and I still get the problem. Using Sonoff basic to control the lights of my home using a wall switch on GPIO14/Ground. Sometimes it takes up to 10 seconds to turn on/off the lights. Anyone knows what can it be?

davidelang, 2017-12-08:
probably because you are having network connectivity issues.

There is a bug in the arduino network library that means that while the device 
is attempting to connect to a mqtt server, nothing at all can happen.