arendst/Tasmota #1069: SwitchMode question

roblad, 2017-10-25:
Hi,

Is it possible to change SwitchMode from 1 that is default to SwitchMode to value 2 somewhere in config or source code ? I need SwitchMode 2 by default.

Thanks
RL

davidelang, 2017-10-25:
see user_config.h

roblad, 2017-10-25:
ok,

what is a parameter ?

jhlweb, 2017-10-25:
around line 154
`#define SWITCH_MODE            TOGGLE            // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, 
PUSHBUTTON, PUSHBUTTON_INV, PUSHBUTTONHOLD or PUSHBUTTONHOLD_INV (the wall switch state)`

roblad, 2017-10-26:
Hi

Smitchmoe1 2
Switchmode2 2

through conslole MQTT solved that, but haw to do it in config, there is a question ?

Regards,
RL

roblad, 2017-10-26:
SWITCH_MODE1 TOGGLE
SWITCH_MODE2 TOGGLE

???

jhlweb, 2017-10-26:
@roblad, maybe your should type a bit more text with your comments to make clear what you want.

closed 2018-05-07 by stale[bot]