arendst/Tasmota #1070: Help with Mac OS

aferraz, 2017-10-25:
Hi all,
 I tried to compile Tasmota on Mac OS High Sierra but get this error:
Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/esp8266_peri.h:24:21: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
                     ^
compilation terminated.

I followed the wiki and installing via Platform Manager esp8266 but nothing.

Someone can help me ?

Thanks

jeylites, 2017-10-26:
Give Arduino IDE a try to see if it works for you.

aferraz, 2017-10-26:
Sorry, I talk about Arduino IDE

jeylites, 2017-10-26:
By the looks of it, you seem to be missing "c_types.h" file from your directory. Copy attached file into your directory. Also be aware on the High Sierra issue. 

[957](https://github.com/arendst/Sonoff-Tasmota/issues/957)

[c_types.h.zip](https://github.com/arendst/Sonoff-Tasmota/files/1417254/c_types.h.zip)

aferraz, 2017-10-27:
Hi @jeylites thank you for your help but after c_types.h there are more files missing.
I think there is a problem installing esp8266 with Arduino board manager.
I switch to PlatformiO and work fine.

jeylites, 2017-10-28:
That’s great it’s working for you on platform io. On a second thought, I ran into similar problem in the past, it just didn’t occur to me when it was first mentioned. I solved it by copying the flies directly from Git to the right directory which worked right after. In the past  I was using Platform io and it was great until an upgrade  which led me to compiling errors. Since than, I been using Arduino IDE.

aferraz, 2017-10-30:
@jeylites  I'll try your suggestion.

closed 2017-10-30 by aferraz