arendst/Tasmota #1071: please add a "TeleRetain" option

qorron, 2017-10-25:
if set, it would set the retain flag on telemetry messages.
motivation:
I'm writing a series of munin and nagios/icinga plugins that would use that data.

those scripts are just executed once and expected to terminate immediately with a result printed to stdout.
so, no time to wait for a telemetry message to show up.
the only way to work around this would be to write some sort of always running programm that lurks on the mqtt, stores the information and hands it over to a monitoring script once it is queried.

with that information retained, the script would connect to the mqtt, reads the retained information, converts it, hands it over to the monitoring system and terminates.

telemetry info could be used to monitor wifi RSSI to get a more or less realtime monitoring of your wifi coverage. e.g. create a heat map.
I'd also monitor voltages to perhaps predict PSU failure on a sonoff module.

outline: ~40 tasmota devices, ~30 of them sonoff touch.

if you think what I'm trying to do is stupid and there is a better solution, please feel free to point that out.

meanwhile, I write that for the sensor output.

ionciubotaru, 2017-10-26:
This is a bad idea. If telemetry messages are retained and you take a decision based on them, if the sonoff goes down you still get the tele message on subscription and take a wrong decision.

Better save the messages and the received time to a database and use them for your future decisions.

qorron, 2017-10-26:
telemetry messages, just like sensor messages, have a timestamp already on them.
e.g.
`tasmota/th16_dachboden/tele/STATE = {"Time":"2017-10-26T09:03:39", "Uptime":4, "Vcc":3.197, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"wald", "RSSI":50, "APMac":"06:18:D6:2B:2C:F8"}}`

actually, this is how I discovered #1072

stefanbode, 2017-10-28:
I also do not understand why there is a retain required. I retain the message and the information in other Programms like ThingWorx or OpenHab. For example the APMac and RSSI. Anhyhow having an option for those people who need it, seems to be ok. It is always the issue to keep RAM requirements in mind. The ESP is limited and arend did a great job not to implement everything.

qorron, 2017-10-28:
the retain flag on published messages does not affect RAM requirements on the ESP as the information is not retained in the ESP but in the MQTT broker. having an option of course would use a few bytes of code on the ESP. having that hardcoded to 1 would not affect the ESPs resources in any way.

I hacked that already to go along with the SensorRetain option to be able to develop stuff without having to wait for this to be implemented. i.e. my SensorRetain also means TeleRetain.

my stuff works already, it autodiscovers every tasmota device connected to the mqtt and graphs RSSI and Vcc values nicely for all devices. it also graphs all sensor data from devices that have sensors on them.

TeleRetain = 1 would also provide one with the ability to see when a device was last seen online and may give a reason why it went offline. e.g. low RSSI or failing PSU. all without any other software just for retaining the data.

stefanbode, 2017-10-28:
The additional RAM and code requirements came from implementing the option in the MQTT command and serial commands. Just hard coding is not the way this great software is working. Also you need some bytes for the description text in the language file.

qorron, 2018-04-24:
I added a TELE_RETAIN switch to the user_config.h in my local git and drag that along ever since.

as can be clearly seen, no additional RAM is wasted and there isn't even anything that needs translation.

`diff --git a/sonoff/sonoff.ino b/sonoff/sonoff.ino
index 71aeb90..8abe590 100644
--- a/sonoff/sonoff.ino
+++ b/sonoff/sonoff.ino
@@ -1885,7 +1885,7 @@ void PerformEverySecond()
 
       mqtt_data[0] = '\0';
       MqttShowState();
-      MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_STATE));
+      MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_STATE), TELE_RETAIN);
 
       mqtt_data[0] = '\0';
       if (MqttShowSensor()) {
diff --git a/sonoff/user_config.h b/sonoff/user_config.h
index 3de3e8f..8b9e027 100644
--- a/sonoff/user_config.h
+++ b/sonoff/user_config.h
@@ -221,6 +221,8 @@
 
 //#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)
 
+#define TELE_RETAIN    1                       // set retain bit on telemetry messages
+
 /*********************************************************************************************\
  * No user configurable items below
 \*********************************************************************************************/`

ascillato, 2018-04-25:
Hi, please add a Pull Request with this to be discussed there, because this is an enhacement. Thanks

arendst, 2018-04-25:
If this fix is all you need I'll add it as designed here; a compile option only.

qorron, 2018-04-26:
yes it's all I need, thank you very much!

closed 2018-04-26 by qorron