arendst/Tasmota #1072: Sonoff SC Modules stopped showing any sensor data

qorron, 2017-10-26:
It appears that my 2 SCs no longer report any sensor data, web page or mqtt.

debug log shows:
`02:37:10 SER: Received AT+STATUS?
02:37:10 SER: Transmit AT+STATUS=4
02:37:13 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
02:37:13 SER: Transmit AT+SEND=fail
02:37:16 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
02:37:16 SER: Transmit AT+SEND=fail
02:37:20 SER: Received AT+UPDATE="humidity":,"temperature":,"light":,"noise":,"dusty":
02:37:20 SER: Transmit AT+SEND=fail`

I'm pretty sure the jumpers on the pcb are back in place since they have worked before and I don't remember opening them a second time. I can't check because I'm 150km away from them for the next days.

what I suspected as a cause was the disabling of some stuff I do not use in the user_config.h but I reenabled everything I had disabled (except for IR-remote and Domoticz)
is there anything from the user_config.h that the SC needs to read sensor data?

the problem appeared first (i.e. the last retained sensor messages of those 2 devices coincide with the night I initially developed the tasmota_prober that blinks unconfigured switches and prompts for a name and some sort of mqtt_autoconfig that lays out a predefined config on an arbitrary number of tasmota devices and also upgrades the firmware if required)
having this in mind, I tried to reset one of them to the defaults in user_config.h but that didn't help.

I'm pretty sure that I caused the problem but nevertheless I'm stuck. :-/

arendst, 2017-10-26:
I had a similar experience caused by an overloaded (crappy) USB power supply.

Also keep in mind the SC uses the serial interface to talk to the Atmel sensor chip so don't try to fool around with those GPIOs.

qorron, 2017-10-26:
I asked the on-site-folk to powercycle one of the SCs and it started working again.
could the amtel be disturbed by a firmware upgrade, a change of the module number (even SC to SC)?

power should be good, they are connected to a seperate high power Anker 5-port USB charger each. no devices were connected or disconnected at the time when the both failed.

is there a way to reset the amtel remotely? or automatically?

closed 2018-05-07 by stale[bot]