arendst/Tasmota #1073: Feature Request: Xenon touch wifi light switch

kezzab25, 2017-10-26:
Has anyone had experience with the Xenon touch switches? I have just ordered one and apparently they are based on the ESP8266 chipset. Will it be as easy as saying they are a Sonoff 3ch light switch?

ionciubotaru, 2017-10-27:
I assume this is the device: https://it.aliexpress.com/store/product/Xenon-US-Standard-Touch-Switch-work-with-amazon-Echo-Alexa-mobile-Remote-control-light-switch-smart/2415037_32814869472.html

Please read https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

kezzab25, 2017-10-27:
Cheers. I know this is probably a dumb question but how do I keep my user_config and platformio files while doing an update from github?

ionciubotaru, 2017-10-27:
put changed params in user_config_overite.h than save it to a different location.
after upgrade copy it back

kezzab25, 2017-11-02:
I had a quick look at one and it seems it has another IC controlling the touch switches and I am guessing it uses serial. Is there any way of writing code for this?

kezzab25, 2017-11-05:
I got it to work. How do I submit it to be added into the code?

`{ "Xenon 3CH",      // Xenon 3CH (ESP8266)
    0,0,0,
    GPIO_KEY2,        // GPIO03 Serial TXD and Optional sensor
    GPIO_REL2,        // GPIO04 Relay 2
    GPIO_KEY3,        // GPIO05 Input 2
    0, 0, 0,          // Flash connection
    0,
    0,
    0,
    GPIO_KEY1,        // GPIO12 Key input
    GPIO_REL1,        // GPIO13 Relay 1
    0,
    GPIO_REL3,        // GPIO14 Relay 3
    0, 0
  }`

enesbcs, 2017-12-02:
@kezzab25 
Could you please upload a photo of the circuit. I am curious. :) Is it really serial MCU controlled, or directly with ESP GPIO?

kezzab25, 2017-12-02:
It's directly controlled via GPIO. I'll have to see if I have some photos

CZ1979, 2017-12-17:
hi kezzab25,
how did you flash the TYWE3S chip inside the xenon wall. 
I bought this switch (from Jinvoo, which is an Xenon oem) an tried to flash it with tasmota firmware.  I had no luck so far! I couldn't get any signal via TX/RX. Also after grounding GPIO0.

Here's a picture of the chip inside.
[image]