arendst/Tasmota #1076: location of gpio4 & 14

edit21, 2017-10-27:
Hi all, i see in the code for sonoff dual and touch (as well as selecting the module type) that these pins should be accessible though i cant find them on the circuit boards  (searched everywhere inc the tech schematics).
Does anyone have any info, would love to update the GPIO location page in the wiki :)

jetema, 2017-10-27:
Have a look at --> https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations

TUISTERa, 2017-10-27:
Nice answer jetema, but both devices are missing from the link that you posted. Please, next time, if you have nothing to say, just dont say nothing

jetema, 2017-10-27:
Ups sorry about the above link, didn't realized those specific devices were not there, hope it helps
This pic of 8266 with GPIO 4 & 14, if you follow the pins/tracks on the board you' ll find the easiest point to connect 
[image]
The full picture is on
https://i.pinimg.com/originals/3b/1b/d7/3b1bd7af3b7db1fba48086c1f558a4d8.png

pascalsaul, 2017-10-27:
https://github.com/arendst/Sonoff-Tasmota/issues/797

My Dual and the extra GPIO's ;)

edit21, 2017-10-28:
That's great info! Can I get rights to edit the Wiki, can help update.

I now have the dual sensor working, just need to get the 4ch pro sensor working now - only have the options for  GPIO 1,2 & 3 though non are listed as a sensor port, is there a way to add a sensor here?

closed 2017-10-28 by edit21